OPCFoundation/UA-.NETStandard #491: Model Compiler Input

sntechno, 2018-06-28:
In the readme for the Model Compiler available at [https://github.com/OPCFoundation/UA-ModelCompiler](url), it is stated that the model compiler will take an XML file following the proper schema for an input. What is the procedure to implement an appropriate XML file into the compiler?

opcfoundation-org, 2018-06-28:
Try:
https://github.com/OPCFoundation/UA-ModelCompiler/blob/master/ModelCompiler/UA%20Model%20Design.xsd

sntechno, 2018-06-28:
Thank you, but I think my question may have been vague. I have followed that schema for the XML file I created, which I intend to input into the tool. Specifically, what is the procedure to implement this XML file which I have already written in the correct format into the tool?

fchaxel, 2018-07-04:
Hi @sntechno 

Not sure it's the question, but if you want to adds/shows nodes into a C# UA server nodes spaces this is a response:

After the xml file compilation you gets 2 interesting files : a .cs and a .uanodes. The .cs must be added to your project as a normal classes file. The .uanodes can be added as an embedded resource.

In the CustomNodeManager2 child class, the namespace referenced in your xml file should (if diifferent than the application namespace ... try first with the same) in the ctor using the SetNamespaces method.
Then the override method LoadPredefinedNodes can be written quite like this:

`

        protected override NodeStateCollection LoadPredefinedNodes(ISystemContext context)
        {
            NodeStateCollection predefinedNodes = new NodeStateCollection();
            
            Assembly _assembly;
            _assembly = System.Reflection.Assembly.GetExecutingAssembly();
            
            // Assume the resource file name is 'MyNodes.PredefinedNode.uanodes'
            String ResName = this.GetType().Namespace + ".MyNodes.PredefinedNodes.uanodes";

            predefinedNodes.LoadFromBinaryResource(context, ResName, _assembly, true);
            return predefinedNodes;
        }

`

New types, folders, ... defined in the xml file are now into your server address space.

Bye.

closed 2018-10-25 by AlinMoldovean