OPCFoundation/UA-.NETStandard #497: OPCUAHDA

vrushalikatkade, 2018-07-10:
Dear Support,

                    I review your application of OPCUAHDA  from "C:\Users\vrushali\Downloads\UA-.NETStandard-master\UA-.NETStandard-master\SampleApplications\Workshop\HistoricalAccess\Client" .
 i have read the ReadHistoryDlg.cs form from HistoricalAccess application. 
[image]

read type processed not working properly it throws exception as follow
[image]

 if you set the read type as readAtTime() this function does not contain any thing
[image]

         please can your share me the proper example with all function of OPCUAHDA

Thanks,
vrushali

vrushalikatkade, 2018-07-11:
plz review this question

AlinMoldovean, 2018-07-26:
Hell @vrushalikatkade ,

You need to make a debug inside the server and find the reason of the BadNotSupported error.
It seems that the sever does not support the Interpolative aggregate for the HistoryRead service.

vrushalikatkade, 2018-07-30:
Thanks for reply,what about readAtTime() this function does not contain any thing

mregen, 2018-10-25:
HDA sample needs to be retested.

closed 2019-02-21 by FischerSeb

FischerSeb, 2019-02-21:
no more reaction. Therefore we close this one.