OPCFoundation/UA-.NETStandard #498: Exceptions in NetCoreConsoleServer' log

yachtingchen, 2018-07-11:
Hi,

I run NetCoreConsoleServer by "dotnet run --project NetCoreConsoleServer.csproj -a" 
and NetCoreConsoleClient by "dotnet run --project NetCoreConsoleClient.csproj -a"

Is it normal to have the following exceptions in the server's log?
Thanks

************************* Logging started at 07/11/2018 14:29:55
7/11/2018 14:29:55.911 Checking application instance certificate.
7/11/2018 14:29:55.917 Checking application instance certificate. CN=UA Core Sample Server, C=US, S=Arizona, O=OPC Foundation, DC=jerry-ks-nb
7/11/2018 14:29:56.171 Checking domains in certificate. CN=UA Core Sample Server, C=US, S=Arizona, O=OPC Foundation, DC=jerry-ks-nb
7/11/2018 14:29:56.506 The server is starting.
7/11/2018 14:29:58.091 Server: Session Monitor Thread Started.
7/11/2018 14:29:58.095 Server: Publish Subscriptions Thread Started.
7/11/2018 14:29:59.131 Channel 0 in Connecting state.
7/11/2018 14:29:59.208 Channel 0: SendHelloMessage()
7/11/2018 14:29:59.222 Channel 0: ProcessAcknowledgeMessage()
7/11/2018 14:29:59.223 Channel 0 in Opening state.
7/11/2018 14:29:59.226 Token #0 created. CreatedAt = 06:29:59.225 . Lifetime = 3600000
7/11/2018 14:29:59.249 Channel 0: ProcessOpenSecureChannelResponse()
7/11/2018 14:29:59.256 Security Policy: http://opcfoundation.org/UA/SecurityPolicy#None
7/11/2018 14:29:59.256 Sender Certificate: (none)
7/11/2018 14:29:59.269 SECURE CHANNEL CREATED [.NetStandard ServerChannel UA-TCP 1.04.353UA-TCP 1.04.353] [ID=26] Connected To: opc.tcp://localhost:4840/ [None/None/Binary]
7/11/2018 14:29:59.272 Token #1 activated. CreatedAt = 06:29:59.225 . Lifetime = 3600000
7/11/2018 14:29:59.273 Channel 26 in Open state.
7/11/2018 14:29:59.341 Channel 26: Close
7/11/2018 14:29:59.342 Channel 26 in Closing state.
7/11/2018 14:29:59.344 Channel 26: SendCloseSecureChannelRequest()
7/11/2018 14:29:59.349 Channel 0 in Closed state.
7/11/2018 14:29:59.350 CLIENTCHANNEL SOCKET CLOSED: 025468F2, ChannelId=26
7/11/2018 14:29:59.357 Channel 0 in Connecting state.
7/11/2018 14:29:59.360 Channel 0: SendHelloMessage()
7/11/2018 14:29:59.361 Channel 0: ProcessAcknowledgeMessage()
7/11/2018 14:29:59.361 Channel 0 in Opening state.
7/11/2018 14:29:59.362 Token #0 created. CreatedAt = 06:29:59.362 . Lifetime = 3600000
7/11/2018 14:29:59.389 Channel 0: ProcessErrorMessage(BadSecurityChecksFailed 'Error received from remote host: ')
7/11/2018 14:29:59.391 Channel 0 in Closed state.
7/11/2018 14:29:59.392 CLIENTCHANNEL SOCKET CLOSED: 02281665, ChannelId=0
7/11/2018 14:29:59.393 RegisterServer2 failed for at: opc.tcp://localhost:4840/. **Exception**=BadSecureChannelClosed
7/11/2018 14:29:59.395 ***EXCEPTION*** BadSecurityChecksFailed 'Error received from remote host: '
7/11/2018 14:29:59.395 Channel 0 in Connecting state.
7/11/2018 14:29:59.434 Channel 0: SendHelloMessage()
7/11/2018 14:29:59.435 Channel 0: ProcessAcknowledgeMessage()
7/11/2018 14:29:59.435 Channel 0 in Opening state.
7/11/2018 14:29:59.436 Token #0 created. CreatedAt = 06:29:59.436 . Lifetime = 3600000
7/11/2018 14:29:59.447 Channel 0: ProcessErrorMessage(BadSecurityChecksFailed 'Error received from remote host: ')
7/11/2018 14:29:59.448 Channel 0 in Closed state.
7/11/2018 14:29:59.449 CLIENTCHANNEL SOCKET CLOSED: 01AC495C, ChannelId=0
7/11/2018 14:29:59.450 RegisterServer failed for at: opc.tcp://localhost:4840/. **Exception**=BadSecureChannelClosed
7/11/2018 14:29:59.452 Register server failed. Trying again in 2000 ms
7/11/2018 14:29:59.452 ***EXCEPTION*** BadSecurityChecksFailed 'Error received from remote host: '
7/11/2018 14:30:01.466 Channel 0 in Connecting state.

mregen, 2018-07-12:
@yachtingchen if you are using the LDS registration you need to copy the server cert to the LDS cert store.

yachtingchen, 2018-07-12:
@mregen Yes, I'm using LDS registration. The exception disappeared after copying the certs. Thanks. 

For anyone who encounter the same problem, you can find the certificate store information of the LDS in the configuration file C:\ProgramData\OPC Foundation\Config\Opc.Ua.DiscoveryServer.Config.xml and the ini file C:\ProgramData\OPC Foundation\UA\Discovery\ualds.ini

closed 2018-07-18 by mregen