OPCFoundation/UA-.NETStandard #499: Port NetCoreConsoleServer to .NET Framework

yachtingchen, 2018-07-11:
Hi,

I tried to port NetCoreConsoleServer to .NET Framework by the following steps:

1. Create a .NET Framework 4.6.1 console application
2. Copy the Program.cs and Opc.Ua.SampleServer.Config.xml into it.

I am able to build and run the application. However, there's an error in the log:

7/11/2018 13:41:57.855 The server is starting.
7/11/2018 13:42:02.942 **Could not load HTTPS Stack Listener. Unable to load DLL 'libuv': The specified module could not be found.** (Exception from HRESULT: 0x8007007E) AggregateException 'One or more errors occurred.'

========================================
Id: Bad
Description: One or more errors occurred.
>>> One or more errors occurred.
---    at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions)
---    at System.Threading.Tasks.Task.Wait(Int32 millisecondsTimeout, CancellationToken cancellationToken)
---    at Microsoft.AspNetCore.Server.Kestrel.Internal.KestrelEngine.Start(Int32 count)
---    at Microsoft.AspNetCore.Server.Kestrel.KestrelServer.Start[TContext](IHttpApplication`1 application)
---    at Microsoft.AspNetCore.Hosting.Internal.WebHost.Start()
---    at Microsoft.AspNetCore.Hosting.HostingAbstractionsWebHostBuilderExtensions.Start(IWebHostBuilder hostBuilder, String[] urls)
---    at Opc.Ua.Bindings.UaHttpsChannelListener.Start()
---    at Opc.Ua.Bindings.UaHttpsChannelListener.Open(Uri baseAddress, TransportListenerSettings settings, ITransportListenerCallback callback)
---    at Opc.Ua.ServerBase.CreateHttpsServiceHost(IDictionary`2 hosts, ApplicationConfiguration configuration, IList`1 baseAddresses, ApplicationDescription serverDescription, List`1 securityPolicies)

>>> **Unable to load DLL 'libuv':** The specified module could not be found. (Exception from HRESULT: 0x8007007E)
---    at Microsoft.AspNetCore.Server.Kestrel.Internal.Networking.Libuv.NativeMethods.uv_loop_size()
---    at Microsoft.AspNetCore.Server.Kestrel.Internal.Networking.UvLoopHandle.Init(Libuv uv)
---    at Microsoft.AspNetCore.Server.Kestrel.Internal.KestrelThread.ThreadStart(Object parameter)
========================================

The error disappeared after I copied the **libuv.dll** from \SampleApplications\Samples\Server.Net4\bin\Debug to the output folder of the application.

I wonder why I need to copy this dll manually. Did I miss any steps for porting it to .NET Framework?
Thanks for your help.

mregen, 2018-07-12:
@yachtingchen foe .Net 4.6 projects there is indeed an additional property in the csproj necessary to tell msbuild which version of the libuv.dll to copy to release:
```
  <PropertyGroup>
    <RuntimeIdentifier>win7-x64</RuntimeIdentifier>
    <BaseNuGetRuntimeIdentifier>win7-x64</BaseNuGetRuntimeIdentifier>
  </PropertyGroup>
```
(adjust for non x64 targets etc.)

yachtingchen, 2018-07-12:
@mregen Thanks for the information. After adding the additional property in the csproj, msbuild still doesn't copy libuv.dll to the output directory. However, I can copy it manually, so it's not a big problem.