OPCFoundation/UA-.NETStandard #511: Anyone who capure encrypted OPC UA packets through Wireshark?

codekimchi, 2018-08-02:
I've found some way to caputre OPC UA packets. 

https://opcconnect.opcfoundation.org/2017/02/analyzing-opc-ua-communications-with-wireshark/

However it works only for security policy none mode. 

Is there anyone who capure encrypted OPC UA packets through Wireshark?

AlinMoldovean, 2018-08-02:
Hello @codekimchi ,

You can also capture the encrypted packets with Wireshark.
But for decrypting the data you'll need to have the private keys of the application certificates and to manually decrypt the packets.

A much easier way to intercept an OPC UA communication between a client and a server (for endpoints without security) is by using the [OPC UA Compliance Test Tool](https://opcfoundation.org/developer-tools/certification-test-tools/opc-ua-compliance-test-tool-uactt/) from OPC Foundation.

codekimchi, 2018-08-08:
@AlinMoldovean How could I use the CTT to capture network packet for debugging purpose.

AlinMoldovean, 2018-08-08:
You need to configure a Client project.
The tool works as a proxy between server and client.

[image]

codekimchi, 2018-08-09:
I follow your guide. However I got the unexpected result. see the below

[image]

[image]

[image]

AlinMoldovean, 2018-08-09:
@codekimchi ,

The endpoint exposed by your server is with security (SignAndEncrypt).
So you will not be able to intercept the communication because CTT works only for endpoints with no security, i.e "None".

I cannot further provide support for configuring and running that tool.

closed 2018-08-09 by AlinMoldovean