OPCFoundation/UA-.NETStandard #512: Time for connection status

famalicom, 2018-08-02:
Hello guys,

I am testing the library and it happens that the time is sometimes wrong according to the status of the connection.
Only the Disconnect status displays the correct time because it uses the computer time, when server time is used (connected and reconnecting), the time is wrong (minus one hour in my case), although the server time is correct.

Thanks, best regards

famalicom, 2018-08-06:
Hi all,

Yes, I use the library as it is in guithub.

I use this:
`UpdateStatus(false, e.CurrentTime, "Connected [{0}]", session.Endpoint.EndpointUrl);`

and in function UpdateStatus I use this:
`m_StatusUpateTimeLB.Text = time.ToLocalTime().ToString("hh:mm:ss");`

but if I change to:
`UpdateStatus(false, e.CurrentTime.ToLocalTime(), "Connected [{0}]", session.Endpoint.EndpointUrl);`

and in function UpdateStatus to:
`m_StatusUpateTimeLB.Text = time.ToString("hh:mm:ss");`

shows the time properly, what's the difference?

Thanks in advance.

AStove, 2018-09-25:
A `DateTime` object normally is set to _local_ but if you use `DateTime.UtcNow` it's set to _utc_. `toString()` will print it without looking what kind of time it is. What `ToLocalTime()` does is change it to _local_ and apply the offset.

UpdateStatus is called 4 times, only in the 'Disconnected' case it returns a UTC time.

When you change the code to `m_StatusUpateTimeLB.Text = time.ToString("hh:mm:ss");` it will print the the 'Disconnected' message in UTC using the client's machine time. The other messages (Communication Error, Reconnecting and Connected) will print in local time using the server time. This could compensate for errors in the time that is set but it's not correct because you are sometimes printing in utc and sometimes in local time.

I suspect either your server or client machine is set to the wrong time and or timezone. If still needed, can you give us the following: 
- Current time and timezone of server.
- Current time and timezone of client.
- Current UTC time.

Keep in mind if you are in timezone UTC+1 and DST is applied, you are actually in zone UTC+2 but Windows will still call it UTC+1.

closed 2018-10-25 by AlinMoldovean