OPCFoundation/UA-.NETStandard #514: What is the editor of the UA-.NETStandard library

codekimchi, 2018-08-06:
I've found the '\Stack\Generated' folder and got so many XML files here.
It must be the generated files by a tool. 
What is the tool? Where could I get that and How to run it?

AlinMoldovean, 2018-08-06:
Hi @codekimchi ,

Those files are imported from the [UA-NodeSet ](https://github.com/OPCFoundation/UA-Nodeset/tree/master/Schema) repository and are generated using [UA-ModelCompiler](https://github.com/OPCFoundation/UA-ModelCompiler) tool.

codekimchi, 2018-08-07:
@AlinMoldovean The XML files I mentioned above are UA-NodeSet files. Right? And UA-ModelCompiler compile these files into source code. However How do you edit the UA-NodeSet files. I mean which tools can I  add or remove node and modify attributes? 

Editing XML files is nonsence.

AlinMoldovean, 2018-08-08:
Yes, those are UANodeSet files.
ModelDesigner generates both NodeSet files and source code from input files in a UAModeDesign format.
There are some commercial tools on the market for generating or editing NodeSet files from vendors like Unified Automation or Softing.

closed 2018-10-25 by AlinMoldovean