OPCFoundation/UA-.NETStandard #518: UnauthorizedAccessException while creating certificate

solcatko, 2018-08-09:
Hi, I have a problem with running the sample on Xamarin iOS, everytime I try to connect I am getting this exception: 

[0:] 8/9/2018 13:10:41.305 Loading application configuration file. /private/var/containers/Bundle/Application/4A445094-1D00-4C14-9EB3-AF169D466A98/XamarinClient.iOS.app/Opc.Ua.SampleClient.Config.xml
[0:] 
************************* Logging started at 08/09/2018 13:10:41
[0:] 8/9/2018 13:10:41.430 Checking application instance certificate.
[0:] 8/9/2018 13:10:41.440 Creating application instance certificate.
[0:] 8/9/2018 13:10:41.459 Deleting application instance certificate.
Unhandled Exception:

System.UnauthorizedAccessException: <Timeout exceeded getting exception details>

I am really not sure how to solve it. Could you please help me?
Thanks

christhiel, 2019-07-18:
I´ve got the same issue while debugging to a physical device(iPad Mini). 
The exception occurs in Utils.GetAbsoluteDirectoryPath(...) at Line 747 when 'Directory.CreateDirectory' is called.

christhiel, 2019-07-20:
I´ve changed the StorePath-Property of TrustedIssuerCertificate and ApplicationCertificate to `Environment.SpecialFolder.MyDocuments` --- Now, it works on iPad.

christhiel, 2019-07-21:
Hi @mregen 
I think, the problem is that the Opc.Ua Assemblies are trying to create new Directories in `/private/var/containers/ **Bundle** /Application/...` with the relative path from the config.xml file.
First, it seems like that is the wrong place and second, they have no access to this directory.
The correct directory is `/var/mobile/Containers/ **Data** /Application/` and the new Directories should be created in `Ua Xamarin Client.iOS PathService`. I´ve created the CertificateStore Directories there like this:

```csharp
public string ApplicationCertificateStorePath
{
     get
     {
           return GetDirectory(Path.Combine(GetCertificateStorePath(), "own"));
     }
}

private static string GetCertificateStorePath()
{
     var appDataPath = Environment.GetFolderPath(Environment.SpecialFolder.LocalApplicationData);
     var certStorePath = Path.Combine(appDataPath, "Opc Foundation/pki");
     return certStorePath;
}

private static string GetDirectory(string path)
{
     if (!Directory.Exists(path))
     {
           Directory.CreateDirectory(path);
     }

     return path;
}
```

And I changed the StorePath-Properties in `Ua Xamarin Client SampleClient.cs` in the CreateCertificate() method:

```csharp
if (Device.RuntimePlatform == "Android")
{
     ...
}
else if (Device.RuntimePlatform == Device.iOS)
{
     // load the application configuration.
     config = await application.LoadApplicationConfiguration(false);
     
     application.ApplicationConfiguration.SecurityConfiguration.ApplicationCertificate.StorePath = DependencyService.Get<IPathService>().ApplicationCertificateStorePath;
     application.ApplicationConfiguration.SecurityConfiguration.TrustedIssuerCertificates.StorePath = DependencyService.Get<IPathService>().TrustedIssuerCertificatesStorePath;
     application.ApplicationConfiguration.SecurityConfiguration.TrustedPeerCertificates.StorePath = DependencyService.Get<IPathService>().TrustedPeerCertificatesStorePath;
     application.ApplicationConfiguration.SecurityConfiguration.RejectedCertificateStore.StorePath = DependencyService.Get<IPathService>().RejectedCertificateStorePath;
}
else
{
     // load the application configuration.
     config = await application.LoadApplicationConfiguration(false);
}
```
And now, it works on a physical device.

closed 2020-03-30 by AlinMoldovean