OPCFoundation/UA-.NETStandard #525: Access MonitoredItems / Subscriptions on server side

tschombe, 2018-08-24:
I have a question regarding server development:

I have a huge amount of datapoints in my server that I need to poll.
But I only want to poll the values of datapoints that clients are currently requesting.

So I am looking for a way on the server to access the currently existing client subscriptions 
or monitored item lists, so that I can only poll the nodes that are registered there.

Is there any sample on how to do this?

AlinMoldovean, 2018-08-24:
Hi @tschombe ,

An option is to implement overrides for OnMonitoredItemCreated and OnMonitoredItemDeleted and keep a local list of monitored nodes.

Examples can be found in DataAccess Server and HistoricalAccess Server.

tschombe, 2018-08-27:
Thanks very much for that hint, this will help me create my own list of monitored nodes

closed 2018-08-27 by tschombe