OPCFoundation/UA-.NETStandard #527: An attempt was made to load a program with an incorrect format libuv.dll

kchtun, 2018-08-28:
Hi all,

I am implementing an OPC UA Server with the last OPC Foundation stack release . The server should support HTTPS for security reasons. I have two executable versions with different target platform configuration (x86 and x64). The x64 version works well (the server hosts the URI properly and the OPC UA client can connect successfully). However, the x86 version cannot host the URI.

After debugging, I localized the problem:
StandardServer(class) -> CreateHttpsServiceHost(method )->listener.Open(uri.Uri,settings,GetEndpointInstance(this));. At this line it catches and give me an error "An attempt was made to load a program with an incorrect format". 

After some Google and some tests I figured out that the root cause of the problem could be the target architecture of the libuv.dll built mode.

PS: Libuv.dll was imported from the sample applications folder in this repository

I need to know:
Is my reasoning right ? If yes, could you suggest a workaround process or provide me with libuv.dll that supports x86 executable ?

Cheers.

AlinMoldovean, 2018-09-11:
Hi @kchtun ,

I think Libuv.dll was downloaded as a package when compiling the project.
Maybe you can try compiling the code in VS with "Any CPU" and check if it works.

kchtun, 2018-09-12:
I downloaded the libuv source code and then re-build it with x86 target platform and now it works.
Thank you BTW.

closed 2018-10-04 by AlinMoldovean