OPCFoundation/UA-.NETStandard #531: Sending large Imagefiles (<1GB)

NichtEuler, 2018-09-13:
Hi,

I have written a client, that can send small (somewhat around 500kB ) imagefiles.
Going above that limit results in the exception 
"BadEncodingLimitsExceeded 'MaxByteStringLength 1048576 < 1651395"
Is OPC UA even meant for such large files? 
And if so, how can the limit be raised?

AlinMoldovean, 2018-09-13:
Hi @NichtEuler ,

The message size in OPC UA is limited for some reasons.
Anyway, it is possible to transfer large files using a"File transfer" mechanism described in specification document "OPC Unified Architecture, Part 5" at Annex C. It requires implementation of a state machine to transfer the file in multiple pieces.

NichtEuler, 2018-09-13:
Hi @AlinMoldovean ,

This is exactly what i am looking for.
Thank you.

closed 2018-09-13 by NichtEuler