OPCFoundation/UA-.NETStandard #534: Type chooser

benjamin1988, 2018-09-14:
Hello,

we have the problem that we can't find a way to transfer a type information. We do have a method with an array of values. Each value has a id an address and at least something like a type. We already tried to use a variant parameter where we can choose the type. But there we also have to set a value. 
Is there any other standard way to only set the TypeInformation? 
Thanks a lot.

NichtEuler, 2018-09-18:
I don't know if this is what you are looking for, but this returns the System.Type

````
        private Type GetSystemType(ExpandedNodeId expandedNodeId)
        {
            VariableNode node = m_session.NodeCache.Find(expandedNodeId) as VariableNode;
            BuiltInType builtInType = TypeInfo.GetBuiltInType(node.DataType);
            Type type = TypeInfo.GetSystemType(builtInType, node.ValueRank);
            return type;
        }
````

Which I followed with 
```
dynamic value= null;
Type systemType = GetSystemType(expandedNodeId);
value = Convert.ChangeType(msg, systemType);
```

Hope this helps.

closed 2019-02-21 by FischerSeb

FischerSeb, 2019-02-21:
See last comment for possible solution