OPCFoundation/UA-.NETStandard #535: Creating an ExtensionObject with Byte[]

AStove, 2018-09-14:
When creating an ExtensionObject with a byte[] in the constructor the compiler implicitly casts byte[] to ExpandedNodeId. It then runs the wrong overloaded constructor of ExtensionObject. (This later on caused BadDataencodingUnsupported for me.):

```
Byte[]  exampleByteArray;
example.Value.WrappedValue = new ExtensionObject(exampleByteArray);
```
Workaround:
```
Byte[]  exampleByteArray;
var eo = new ExtensionObject();
eo.Body = exampleByteArray;
example.Value.WrappedValue = eo;
```

Anyone have a better idea? Could this be improved?

AlinMoldovean, 2018-09-25:
Hello @AStove ,

The problem will be the same for other types like _String, Guid, uint, NodeId_ 
The correct way to set the value would be:

```
Byte[]  exampleByteArray = new byte[] { 0x20, 0x20, 0x20, 0x20, 0x20, 0x20, 0x20 };
example.Value.Value= exampleByteArray;
```
or

```
Byte[]  exampleByteArray = new byte[] { 0x20, 0x20, 0x20, 0x20, 0x20, 0x20, 0x20 };
example.Value.WrappedValue = new Variant(exampleByteArray);
```

So, PR #536 will be closed now

closed 2018-09-25 by AlinMoldovean

AStove, 2018-09-25:
That fixes it, thanks for the help.