OPCFoundation/UA-.NETStandard #540: DataTypes sample problem

trofimich, 2018-09-20:
I've opened node ParkingLot -> DriverOfTheMonth -> PrimaryVehicle and see an error: BadUnexpectedError (0x80010000)

Look at screenshot here: 
[image]

AStove, 2018-09-20:
Can you provide a stacktrace or entries from the logging?

trofimich, 2018-09-20:
Hello!
I did not undertand what i need to do. I simply compiled and ran "DataTypes Server" project from "UA Quickstart Applications" solution and connected to it with UaExpert. I've got no errors in application. And you can see log of UaExpert - nothing interesting there...

I've expected to have possibility to set all fields of Vehicle structure as a single value direcly in UaExpert, but not found such possibility. Instead of that i found this error and decided to write about it...

AStove, 2018-09-20:
The software Unified Automation provides is a closed source C++ implementation based on the OPC UA standard. The error you provide comes from their software so ask them. The DataTypes Server might still be at fault but there's no way to know without more info.

The tag you are trying to read is of the type "VehicleType" the UaExpert client doesn't know how to display the value of that type. That does not seem too strange.

trofimich, 2018-09-21:
The problem is not only in UaExpert. Same problem in Quickstart DataTypes Client. Look at screenshot...

In general, i don't understand what does this sample do... It would be nice to add some description where to look and what to do.
[image]
 Because currently sample do something, but it's not clear what exactly and what sence of all this...

AStove, 2018-09-24:
It looks like the _Quickstart DataTypes Client_ is exactly the same as the _Empty Client_. This does not seem right. This example needs to be modified to at least include the datatypes used in the DataTypes Server.

AStove, 2018-09-24:
Some of the types are declared in the DataTypes Library which both the Server and Client use. Others are only defined in the DataTypes Server project hence why the Client doesn't know about it and gives an exception.

AlinMoldovean, 2018-09-24:
The problem was in the Utils.Clone() method.
A PR containing the fix was created.

trofimich, 2018-09-24:
Hi all!

Yet one two problems with these projects:

1. It's not clear what to do and what to look to understand what benefits data types give. It would be naice to add some comments or description file.

2. Current sample built using ModelCompiler tool. We don't use ModelCompiler in our project and it's completely unclear for us how to add neccessary nodes manually. The proposition is to implement sample in poor C# without any external tool (such as model compiler) usage. I think it's more correct for project which demonstrates basic library possibilities...

PR #543 (2018-09-25): Update in Utils.Clone() method.
Fixes #540. Update in Utils.Clone() method.
commit 5f91c46a (2018-09-25): Fixes #540. Update in Utils.Clone() method.
closed 2018-09-25 by AlinMoldovean