OPCFoundation/UA-.NETStandard #541: Samples on poor C# without using of ModelCompiler tool

trofimich, 2018-09-20:
We need to create address space using external information. So, all samples which use ModelCompiler, are not suitable for us.

For example, let's take DataTypes sample. It uses ModelCompiler and we can not understand what steps shoud we do in or codes to add suport of our data types.

fchaxel, 2018-09-26:
\SampleApplications\Workshop\Reference\Server shows how to creates quite all type of nodes in the address space without using the Modelcompiler. 
In particular ReferenceNodeManager.cs contains methods CreateFolder, CreateObject, CreateObjectType, CreateDataItemVariable, ...

tschombe, 2018-09-28:
unfortunately as far as I see the Reference Server does not use or creata any custom type.

I also want to create custom types without using the modelcompiler

trofimich, 2018-09-28:
I found EmptyNodeManager in file ReferenceNodeManager.cs and method CreateDataType, but this method has no usages :-(
[image]

trofimich, 2018-09-28:
The problem is not in node type creation. I can't understand what to do to add encodings... I've already created custom data type and implemented IEncodeable interface, but can't understand how to link ancoding and data type...

tschombe, 2018-09-28:
Yes, I am in the exact same Position,
I see what the model Compiler generates to define a custom data type.
I used Quickstarts.DataTypes.Types.DataTypes.cs as a template (e.g. VehicleType).
I too miss the IEncodable info and 
what I am missing too is the Code that attach this type object in the types Folder of the address space and generate items based on that type.

AlinMoldovean, 2018-09-28:
Hello,

The parts generated by ModelCompiler are :
  - C# code in Classes.cs, Constants.cs and DataTypes.cs
  - NodeSet files in different formats: NodeSet.xml, NodeSet2.xml, PredefinedNodes.xml, PredefinedNodes.uanodes
  
The parts used by a server project (e.g. DataTypes Server, Boiler Server) are:
 - C# generated source code
 - PredefinedNodes.uanodes file

The _.uanodes_ file contains the definition of the address space and is loaded in _LoadPredefinedNodes()_ method

```
 predefinedNodes.LoadFromBinaryResource(context, 
                "Quickstarts.DataTypes.Types.Quickstarts.DataTypes.Types.PredefinedNodes.uanodes", 
                typeof(Quickstarts.DataTypes.Types.VehicleType).Assembly, 
                true);
```
This will create the part of the address space required to correctly expose the types and instances to clients
See #243 , #444 

The list of custom data types implementing the IEncodable interface is loaded in the list of types known by the server using the _Factory.AddEncodeableTypes()_ call in _CreateMasterNodeManager()_

```
            // add the types defined in the quickstart information model library to the factory.
            server.Factory.AddEncodeableTypes(typeof(Quickstarts.DataTypes.Types.VehicleType).Assembly);
            server.Factory.AddEncodeableTypes(this.GetType().Assembly);
```

The process is quite complex and there is no easy way to implement it without ModelCompiler support.

tschombe, 2018-10-01:
ok, so I now try to use the modelcompiler.

I just want to generate an empty custom type with the following XML Definition (ModelDesign.xml):

`
<?xml version="1.0" encoding="utf-8" ?>
<opc:ModelDesign
	xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
	xmlns:xsd="http://www.w3.org/2001/XMLSchema"
	xmlns:opc="http://opcfoundation.org/UA/ModelDesign.xsd"
  xmlns:ua="http://opcfoundation.org/UA/"
  xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd"
	xmlns="http://opcfoundation.org/Quickstarts/Boiler"
  TargetNamespace="http://opcfoundation.org/Quickstarts/Boiler"
>
  
  <opc:Namespaces>
    <opc:Namespace Name="OpcUa" Prefix="Opc.Ua" XmlNamespace="http://opcfoundation.org/UA/2008/02/Types.xsd">http://opcfoundation.org/UA/</opc:Namespace>
    <opc:Namespace Name="Sara" Prefix="Sara.OpcUa.Server">http://testing.schaeffler.com/sara</opc:Namespace>
  </opc:Namespaces>

  
  <opc:ObjectType SymbolicName="SaraBaseDataType" BaseType="ua:BaseObjectType">
    <opc:Description>A generic sara data type representing a system component.</opc:Description>
  </opc:ObjectType> 

</opc:ModelDesign>
`

I then try to call the model Compiler:
opc.ua.modelcompiler.exe -d2 ".\ModelDesign.xml" -cg ".\ModelDesign.csv" -o2 ".\"

and get a null reference exception:
Der Objektverweis wurde nicht auf eine Objektinstanz festgelegt.
bei Opc.Ua.ModelCompiler.ModelGenerator2.WriteTemplate_ConstantsSingleFile(String filePath, List`1 nodes) bei Opc.Ua.ModelCompiler.ModelGenerator2.GenerateMultipleFiles(String filePath, Boolean useXmlInitializers, String[] excludedCategories) bei Opc.Ua.ModelCompiler.Program.ProcessCommandLine2(List`1 tokens)

Any Idea what I am doing wrong?

closed 2018-10-26 by AlinMoldovean