OPCFoundation/UA-.NETStandard #545: Subscription - collecting data in sampling mode

lukaszryszka, 2018-09-25:
Hi everyone,
I am trying to use the subscription mechanism to collect data from the server. I would like to collect the data every certain period of time (let say 1000 ms) even though the monitored value is not being changed. 
I tried to use a following code but it does not work:
```
m_subscription = new Subscription(m_session.DefaultSubscription);
m_subscription.PublishingEnabled=true;
m_subscription.PublishingInterval=1000;
m_session.AddSubscription(m_subscription);
m_subscription.Create();

monitoredItem = new MonitoredItem( m_subscription.DefaultItem );
monitoredItem.StartNodeId = NodeId;
monitoredItem.AttributeId = Attributes.Value;
monitoredItem.MonitoringMode = MonitoringMode.Sampling;
monitoredItem.SamplingInterval = 1000;
monitoredItem.QueueSize =0;
monitoredItem.DiscardOldest = true;
monitoredItem.Notification += new MonitoredItemNotificationEventHandler( monitoredItem_Notification);
m_subscription.AddItem( monitoredItem );
```

awcullen, 2018-09-25:
Try setting the monitoredItem.Filter  to a new DataChangeFilter created with DataChangeTrigger.StatusValueTimestamp.   The default is StatusValue which filters out changes where the only difference is Timestamp.

awcullen, 2018-09-25:
Also, you probably want to change the item's  MonitoringMode to MonitoringMode.Reporting

astrowalker, 2018-09-26:
One missing piece for sure -- `m_subscription.ApplyChanges()` at the end.

fchaxel, 2018-09-26:
Hi @lukaszryszka 

This is not the Opc way of working. Publish interval is the minimum time between call concerning subscription. No data modification ... no notification.
SamplingInterval is the sampling time you want the server try to query the device for a given item : if SamplingInterval = 10 and Publish =1000 then no notification can occure faster than 1 per second (with a maximum of 100 samples if the queue size if big enough, and only if the value is changing).

See http://documentation.unified-automation.com/uasdkc/1.4.0/html/L2UaSubscription.html

If you want the value each 1000ms, when read it in a loop or re-create the periodic sampling time on the client side.

Bye.

lukaszryszka, 2018-09-26:
Hi @astrowalker, it looks like the DataChangeFilter filter does not solve this issue:
```
monitoredItem.MonitoringMode = MonitoringMode.Reporting;
monitoredItem.SamplingInterval = 500;
monitoredItem.QueueSize = 0;
monitoredItem.DiscardOldest = true;
monitoredItem.Filter = new DataChangeFilter()
{
    Trigger = DataChangeTrigger.StatusValueTimestamp,
    DeadbandType = (uint)DeadbandType.Absolute
};
subscription.AddItem(monitoredItem);
subscription.ApplyChanges();
```
I only receive the initial value but not next. A similar issue was posted in the past https://github.com/node-opcua/node-opcua/issues/184

I have noticed that you pointed out an example (https://gist.github.com/coussej/49205e23c4d764d0380f) which works, as I would expect. I only want to ask you for confirmation that in this example you receiving the value every certain period of time.

So far it looks like the only way to read periodically the value from OPC server is using a cyclic read as @fchaxel  and @AlinMoldovean suggested.

lukaszryszka, 2018-09-26:
Sorry, @astrowalker my fault. The answer was addressed to Awcullen. Thank you for answer