OPCFoundation/UA-.NETStandard #546: Is there any solution to import nodeset xml file to opc ua server in C#?

ggokka, 2018-09-27:
I've just found the below information showing the direct instance address space from a nodeset xml file in nodejs.

https://stackoverflow.com/questions/40866428/how-can-i-generate-a-server-addressspace-from-a-xml-nodeset-file/41207337#41207337

I just want to know the equvalent of .NET

NichtEuler, 2018-09-28:
Hi @ggokka 

I looked up this function for a CustomNodeManager2 
```
        private void ImportXml(IDictionary<NodeId, IList<IReference>> externalReferences, string resourcepath)
        {
            NodeStateCollection predefinedNodes = new NodeStateCollection();

            Stream stream = new FileStream(resourcepath, FileMode.Open);
            Opc.Ua.Export.UANodeSet nodeSet = Opc.Ua.Export.UANodeSet.Read(stream);

            SystemContext.NamespaceUris.Append(nodeSet.NamespaceUris.ToString());
            nodeSet.Import(SystemContext, predefinedNodes);

            for (int ii = 0; ii < predefinedNodes.Count; ii++)
            {
                AddPredefinedNode(SystemContext, predefinedNodes[ii]);
            }
            // ensure the reverse refernces exist.
            AddReverseReferences(externalReferences);
        }
```
which you should call like this in the CreateAddressSpace function
```
string resourcepath = "Path/to/xml.xml";
ImportXml(externalReferences, resourcepath);
```

If there is a better way let me know.

fabsmailbox, 2018-10-02:
Hi NichtEuler & ggokka,

thanks for the solution, I had the same question and it helps a lot.
One problem remains:
If I import an OPC UA method with input arguments everything is shown nicely in e.g. UAExpert when browsing.

But when calling the method (and entering the argument values in UAExpert GUI) I get a BadArgumentsMissing return. When debugging it shows the MethodStates.InputArgument is null.

Is the Import functionality limited or am I doing something wrong?
Any hints on that please? 

Thanks a lot
Fabs

ggokka, 2018-10-04:
Hi @fabsmailbox 
Which files have you used to append nodes in the nodeset.xml to your opc ua address spaces?
The .uanodes was important? or not? I'm curious.

ggokka, 2018-10-04:
@fabsmailbox 
Have you checked the #459

NichtEuler, 2018-10-04:
Hi @fabsmailbox,

I don't have any knowledge about methods, because I am just learning how to OPC UA ;)
But the import function can be found at Opc.Ua.Core in the folder Schema.
The file name is UA NodeSetHelpers.cs and the called function 
`private NodeState Import(ISystemContext context, UANode node)`

 Maybe your problem is here:

```
case NodeClass.Method:
                {
                    UAMethod o = (UAMethod)node;
                    MethodState value = new MethodState(null);
                    value.Executable = o.Executable;
                    value.UserExecutable = o.UserExecutable;
                    importedNode = value;
                    break;
                }
```

fabsmailbox, 2018-10-04:
Hi @ggokka and @NichtEuler 

thanks so much for your help. To clarify:

I created a simple object and method with a single input parameter (string) in UAModeler and exported it to XML.
My NodeMangers is derived from CustomNodeManager2.

In the overwritten `CreateAddressSpace `I call the Method `ImportXml `as shown in NichtEulers post above.
As said, in UAExpert everything looks good; I have this object with the method and its input parameters shown as usual.
But when calling it, I get a _BadArgumentsMissing_ 
--> Just a comment: This error message is misleading in this case: The method does expect **zero** arguments, but when calling with UAExpert I actually provide one. Reson: Methodstates Inputarguments is null.

I also tried with overriding the AddBehaviourToPredifinedNodes as shown in [(https://github.com/OPCFoundation/UA-.NETStandard/issues/459#issuecomment-394357808)]
with no success.

Using a method without inputarguments works fine though, my MethodHandler is invoked and returns _GOOD_.

NichtEuler, 2018-10-22:
Hi @codekimchi ,

I imported your XML with my Server and it worked fine.

NichtEuler, 2018-10-23:
Hi @ggokka ,

Add the folder Demo to Opc.Ua.Sample.

[Demo.zip](https://github.com/OPCFoundation/UA-.NETStandard/files/2505611/Demo.zip)

Then add the line 
`nodeManagers.Add(new global::Demo.DemoNodeManager(server, configuration));`
to Opc.Ua.Sample.SampleServer in CreateMasterNodeManager. (Comment out other nodemanagers that are being added if you like)

After you have done that change the Uri in your xml from `http://yourorganisation.org/PlainNode/`
to `http://demo.demo/Demo` as this is the NameSpace this server is working with.

Now start up the SampleServer.

Good luck :)

NaviComputer, 2018-11-22:
Hi,

thank you very much for the demo code. It works fine and I can import a nodeset to the UA server address space. 
However, how can I create the model classes and identifiers from a UA nodeset2 xml? The Modelcompiler only takes TypeModel xmls... Am I missing something?

I have to implement a companion specification which is only available as a Nodeset2.xml (like most UA server specification that I know). Creating the Type Model xml by hand is really not an option as there are hundreds of types in the model.

I really like the OPC foundation .NET SDK but if I'm not wrong it's missing the most important feature...

Thanks and best regards

AlinMoldovean, 2018-11-26:
@NaviComputer 

The standard way to create the C# code for the NodeSet2.xml classes is by using [UA_ModelCompiler](https://github.com/OPCFoundation/UA-ModelCompiler).
This is how most of the companion nodesets are provided on https://github.com/OPCFoundation/UA-Nodeset

NakWarsi, 2019-02-13:
i would like to thank you @NichtEuler for giving such an amazing demo which made everything clear
Thank you

NichtEuler, 2019-02-13:
@NakWarsi Thank you.
I edited the code for multiple namespaces in the nodeset XML-File just now.
If you are looking for hints on how to do something have a look at the `UA Quickstart Applications.sln`.
There are several Examples on Methods, DataTypes etc.

NaviComputer, 2019-02-13:
> @NaviComputer
> The standard way to create the C# code for the NodeSet2.xml classes is by using UA_ModelCompiler.
> This is how most of the companion nodesets are provided on https://github.com/OPCFoundation/UA-Nodeset

sorry, but I don't understand this answer. If I have only the NodeSet2.xml, how to generate the coder? If I'm not wrong the UA_modelCompiler uses a different input format and can only output NodeSet2.xml

NichtEuler, 2019-02-20:
> > @NaviComputer
> The standard way to create the C# code for the NodeSet2.xml classes is by using UA_ModelCompiler.
> This is how most of the companion nodesets are provided on https://github.com/OPCFoundation/UA->Nodeset
>
>@AlinMoldovean
>
>sorry, but I don't understand this answer. If I have only the NodeSet2.xml, how to generate the coder? If >I'm not wrong the UA_modelCompiler uses a different input format and can only output NodeSet2.xml

@AlinMoldovean The UA-Modelcompiler doesn't seem to take NodeSet2.xml.
When trying to use a NodeSet2.xml it looks like [this.](https://github.com/OPCFoundation/UA-ModelCompiler/issues/26) 
Or is it that we are using the wrong syntax?

TindustrialMV, 2020-02-11:
Thank you so much for your demo application. It helped me so much! @NichtEuler

Futtrup, 2022-08-19:
Hi, i also trying grasp the whole "how to OPC-UA".
I have followed this thread and succesfully created an OPC-UA server with the Demo node set.
But when im trying activating all these step on the pre-compiled PackML nodetset2.xml from the nodeset collection, my UaExpert Client only shows the root folder without any content.
Do you have any clues of what i am missing?
My best guess i that the xml nodeset2 file uses more than just base types.

MD-V, 2022-08-19:
@Futtrup which NodeSet are you using? NodeSets can specify instances which are then created in the root but they don't have to.

Futtrup, 2022-08-22:
Hi,

Can you see if i have a mismatch in the NodeSet types i am trying to use and the type in the XML file?

I am trying to activate this pre-compiled NodeSet2.xml file in my sample-server:
https://github.com/OPCFoundation/UA-Nodeset/tree/latest/PackML

And this is my custom node manager class => 

    public class PackMLNodeManager : Opc.Ua.Server.CustomNodeManager2
    {
        #region Constructors
        /// <summary>
        /// Initializes the node manager.
        /// </summary>
        public PackMLNodeManager(IServerInternal server, ApplicationConfiguration configuration)
        :
            base(server, configuration, Namespaces.PackML)
        {
            SystemContext.NodeIdFactory = this;

            m_configuration = configuration.ParseExtension<ReferenceServerConfiguration>();

            if (m_configuration == null)
            {
                m_configuration = new ReferenceServerConfiguration();
            }

        }
        #endregion

        #region INodeIdFactory Members
        /// <summary>
        /// Creates the NodeId for the specified node.
        /// </summary>
        /// <param name="context">The context.</param>
        /// <param name="node">The node.</param>
        /// <returns>The new NodeId.</returns>
        public override NodeId New(ISystemContext context, NodeState node)
        {
            uint id = Utils.IncrementIdentifier(ref m_lastUsedId);
            return new NodeId(id, m_namespaceIndex);
        }
        #endregion

        #region INodeManager Members
        /// <summary>
        /// Does any initialization required before the address space can be used.
        /// </summary>
        /// <remarks>
        /// The externalReferences is an out parameter that allows the node manager to link to nodes
        /// in other node managers. For example, the 'Objects' node is managed by the CoreNodeManager and
        /// should have a reference to the root folder node(s) exposed by this node manager.  
        /// </remarks>
        public override void CreateAddressSpace(IDictionary<NodeId, IList<IReference>> externalReferences)
        {
            lock (Lock)
            {
                IList<IReference> references = null;

                if (!externalReferences.TryGetValue(ObjectIds.ObjectsFolder, out references))
                {
                    externalReferences[ObjectIds.ObjectsFolder] = references = new List<IReference>();
                }

                string resourcepath = "C:\\Users\\kaf\\source\\downloads\\OPC-UA\\UA-.NETStandard-Samples-master\\Samples\\Opc.Ua.Sample\\PackML\\Opc.Ua.PackML.NodeSet2.xml"; 
                ImportXml(externalReferences, resourcepath);                
            }
        }

        /// <summary>
        /// Import NodeSets from xml
        /// </summary>
        /// <param name="path">String to path of XML</param>
        private void ImportXml(IDictionary<NodeId, IList<IReference>> externalReferences, string resourcepath)
        {
            NodeStateCollection predefinedNodes = new NodeStateCollection();

            Stream stream = new FileStream(resourcepath, FileMode.Open);
            Opc.Ua.Export.UANodeSet nodeSet = Opc.Ua.Export.UANodeSet.Read(stream);

            SystemContext.NamespaceUris.Append(nodeSet.NamespaceUris.ToString());
            nodeSet.Import(SystemContext, predefinedNodes);

            for (int ii = 0; ii < predefinedNodes.Count; ii++)
            {
                AddPredefinedNode(SystemContext, predefinedNodes[ii]);
            }
            // ensure the reverse refernces exist.
            AddReverseReferences(externalReferences);
        }
        /// <summary>
        /// Replaces the generic node with a node specific to the model.
        /// </summary>
        protected override NodeState AddBehaviourToPredefinedNode(ISystemContext context, NodeState predefinedNode)
        {
            BaseObjectState passiveNode = predefinedNode as BaseObjectState;

            if (passiveNode == null)
            {
                return predefinedNode;
            }

            NodeId typeId = passiveNode.TypeDefinitionId;

            if (!IsNodeIdInNamespace(typeId) || typeId.IdType != IdType.Numeric)
            {
                return predefinedNode;
            }
            return predefinedNode;
        }

        #endregion
        #region Private Fields
        private ReferenceServerConfiguration m_configuration;
        private List<BaseDataVariableState> m_dynamicNodes;
        private ushort m_namespaceIndex;
        private ushort m_typeNamespaceIndex;
        private long m_lastUsedId;
        #endregion
    }

Futtrup, 2022-08-22:
[DataContract(Namespace = Namespaces.PackML)]
    public class ReferenceServerConfiguration
    {
        #region Constructors
        /// <summary>
        /// The default constructor.
        /// </summary>
        public ReferenceServerConfiguration()
        {
            Initialize();
        }

        /// <summary>
        /// Initializes the object during deserialization.
        /// </summary>
        [OnDeserializing()]
        private void Initialize(StreamingContext context)
        {
            Initialize();
        }

        /// <summary>
        /// Sets private members to default values.
        /// </summary>
        private void Initialize()
        {
        }
        #endregion

        #region Public Properties
        #endregion

        #region Private Members
        #endregion
    }

Futtrup, 2022-08-22:
public static partial class Namespaces
    {
        /// <summary>
        /// The namespace for the nodes provided by the server.
        /// </summary>
        public const string PackML = "http://opcfoundation.org/UA/PackML/";
    }