OPCFoundation/UA-.NETStandard #548: How to retrieve more than one subscribed variable?

jmota7dev, 2018-10-01:
I have been reading the code and some issues from the legacy project and I figured out we can use ReadValues as a function to retrieve all the variable I'm subscribed using list of values and a list of types.

My question is... How can I retrieve every value when I don't know the list of types I'm receiving?
In case of needing the fixed types... How can I built those IList?

Thanks!

astrowalker, 2018-10-01:
I am confused -- when you subscribe to some nodes you don't read values from nodes (because it is pointless), those values are delivered to you by subscription mechanism.

After dequeueing the values from the monitored item you can simply check the type and convert it to more suitable one.

Check out some demo with subscription, it should have all the required pieces you want.

jmota7dev, 2018-10-02:
I'm checking the UWP version SampleClient, and I'm feeling a bit lost following this. 

In order to retrieve the whole list of variables and values that I'm subscribed to... Do I have to retrieve both from the MonitoredItem?

Thanks for your support!

astrowalker, 2018-10-02:
Maybe let's get to basics -- for example how to deal with single node. You subscribe to its change, when it change, the event will fire, in the handler of this event you dequeue the values for this monitored node (please note you cannot assume you will get single value, because there could be already several updates), you can print those new values in console (for example).

So, do you understand that part and you can write a code for it?

jmota7dev, 2018-10-02:
Yes, I was able to get information from one single node but I couldn't retrieve the whole tree to receive all the nodes related to the variables that I'm subscribed to and their values. I don't know if there's a specific method for that.

Thanks for your support!!

astrowalker, 2018-10-02:
In order to get updates from other nodes you need to subscribe to them as well. Please note event will be triggered only for those item which changed.

jmota7dev, 2018-10-02:
I'm taking a look at how to subscribe to different variables at the same time. The OPC client provides a way to subscribe to a single variable clicking on it and pressing in a button (using ContextMenu_OnReport). I'm not sure about how to do it with a batch of variables. 

I don't know if it exists a specific method for that.

astrowalker, 2018-10-03:
How do you add one item to subscription?

          subscription.AddItem(item);

How do you add another one?

          subscription.AddItem(another_item);

And so on. I wonder, since you wrote you subscribed to one item, what problem was here with adding more items?

jmota7dev, 2018-10-03:
Yes, I think my problem is the event I'm using to start subscribing variables.

Do I have a way to subscribe to those events when the connection with the server starts?

astrowalker, 2018-10-03:
`MonitoredItem` has event `Notification`. You should listen to it, and once it is triggered you read the values from the item (you don't subscribe in sense of OPC-UA to this even, at this stage it is C# event mechanism).

astrowalker, 2018-10-03:
When you subscribe to changes in given node you create `MonitoredItem` item, correct? This item contains that event. Simply add a handler for it.

AlinMoldovean, 2018-10-05:
Hello @jmota7dev ,

In OPC UA there is no standard way to get a complete tree structure in a single operation.

I think you are interested to get the "value" inside the tree.
And that is possible, but only for instances of Variable types that expose in the Value attribute a data type containing the same structure as the variable type definition.
One example is the ServerStatus node.

[image]

jmota7dev, 2018-10-15:
Hi! 

After different tries, I have been checking the OPC NetCoreConsoleClient, trying to get a connection with the server. Using a Simulation Server like Matrikon, I'm connecting to the EndPoint but I'm always receiving the error "BadSessionIdInvalid". 

I only change the endpointURL.

Do I need to change anything else?

closed 2020-01-16 by AlinMoldovean