OPCFoundation/UA-.NETStandard #549: Questions regarding subscriptions

Abarbz, 2018-10-02:
Hello,
I am trying to get data from several nodes (5 right now, but the ideal target would be around 50) on an OPC-UA server. The data is refreshed on the server at 50Hz.
Here's a collection of question I was not able to find answer to by looking at sample and existing issues on GitHub.

1. What is the difference between registering an event handler to every monitored item vs using the subscription's FastDataChangeCallback?
I tried doing the latest, to no avail: the registered callback is never called. Is there something special to enable it?

2. With the following client configuration on a simple .Net console application
> Publishing Interval = 200
> Sampling Interval = 10
> Queue Size = 15 ( 10 + safety margin)

I have a strange behavior: notification keeps coming long time after the subscription is stopped with 
```csharp
subscription.PublishingEnabled = false;
subscription.Delete(false);
session.RemoveSubscription(subscription);
```
like something was slowing them down.
For instance, if I let the test run for 10s, I have to wait about the same time after deleting the subscription for the last notification to arrive.
My event handler looks something like
```csharp
private void TransferNewMonitoredItemValue(
              MonitoredItem monitoreditem, 
              MonitoredItemNotificationEventArgs e)
{
   IList<DataValue> values = monitoreditem.DequeueValues();
   foreach (DataValue dequeuedValue in values)
   {
	//simply store dequeuedValue in a collection for further processing
   }
}
```
What could cause such a slow incoming notifications rate?

3. With the same configuration as **2.** , I was expecting to get several data in my list of values obtained through `` monitoreditem.DequeueValues();`` (around 10, since the data is modified at 50H on the server, and I'm only publishing at 5Hz), but I get multiple notification with a single element every time. 
What is the purpose of the collection then?

4. Finally, when comparing the final data, I do miss some of it (the values on the server are basic counter incremented by one 50 times per second, which allows me to detect gaps). 
I tried an equivalent test using the same publishing/sampling/queue values with UAExpert Data Logger View, and it seems that no point is missing.
What am I doing wrong?

The *almost* complete code I am using for those tests can be found [here](https://gist.github.com/Abarbz/05d4081ce4e5a71e1e91804dab8d9706).

Abarbz, 2018-10-10:
If this is not the place for those kind of questions, could you please point me to the right place?

davidemess, 2018-10-12:
Hi @Abarbz ,

I have tried subscription only today, so I can't help so much. But do you remembered to call ApplayChanges() after:

`subscription.PublishingEnabled = false;
subscription.Delete(false);
session.RemoveSubscription(subscription);`

It seems that the client stop to send KeepAlive Subscription but the server waits until timeout (of keepalive) to stop the subscription.

Abarbz, 2018-10-15:
@davidemess indeed I was not calling it, but I tried with a call to `subscription.ApplyChanges();` after my call to `subscription.PublishingEnabled = false;` and it does not change the beavior: it seems it is more related to the time to handle a lot of cached notifications...

Abarbz, 2018-10-15:
After some more testing, I think I figured out the issue explained in the second point: it seems that by default, when in Debug mode with debugger attached, a lot of information are traced.
Either building in release, or without debugger, or calling `Utils.SetTraceOutput(Utils.TraceOutput.Off);` before the subscription greatly improves speed of handling all the notifications.
Is this default behavior intended?

closed 2019-02-21 by FischerSeb

FischerSeb, 2019-02-21:
See last comment. Issue solved