OPCFoundation/UA-.NETStandard #551: GDS and External CA questions.

eoursel, 2018-10-04:
Just something i am not really sure how to configure the GDS. 

- I have a ROOT CA not managed by the GDS. It could be managed by AD Certificates Services or Easy-rsa for instance. Is it possible to configure the GDS as an intermediate CA of this ROOT CA? Is it supported in the current source code or do we have not modify the source code? 
- Another point is the case that the  Intermediate CA of the GDS are completely managed by an external CA (AD CS or something else). Is it really possible this kind of use case?  

Thank you

mregen, 2018-10-26:
@eoursel Hi Eric,
I haven't fully tested the use case but with the current codebase it should be possible to use an intermediate CA, as long as the private key is provided as .pfx in the authorities folder. The configuration steps are manual.
Then the root ca needs to be manually added to the gds trustlist for distribution.
Another problem might be that 'FinishRequest' will not return the whole chain.
For the use case of a completely managed intermediate CA, it depends if you have access to the private key. If this use case requires to forward signing requests to the CA, there is some code enhancement necessary.

eoursel, 2018-10-26:
Thanks @mregen 

The most common use case should be to forward signing requests to the external CA. One of my customers has the project to implement its own PKI with a product like Microsoft AD CS. if we want to connect a GDS which will be hosted on a container we have no other choice to forward signing requests to the real CA. 

Let me know if you agree with this enhancement.

mregen, 2021-01-12:
outdated...

closed 2021-01-12 by mregen