OPCFoundation/UA-.NETStandard #555: Is it possible to create new nodes in the existing node tree in OPC UA Server?

pepeevich, 2018-10-09:
Hello,

Is it possible to create new nodes in the existing node tree in OPC UA Server using OPC UA .NET SDK?

For example, I have the following OPC UA nodes tree on my server:

Objects
    Obj1
         Prop1
         Prop2
     Obj2
         Prop1
         Prop2

Is it possible to add nodes to the tree from the code using OPC UA .NET SDK?

AlinMoldovean, 2018-10-09:
Hello @pepeevich ,

Please take a look at ReferenceServer sample project.
I has examples for creating  all kind of nodes.

pepeevich, 2018-10-09:
> 
> 
> Hello @pepeevich ,
> 
> Please take a look at ReferenceServer sample project.
> I has examples for creating all kind of nodes.

Excuse me, I update my issue description and title. I want to create nodes on the server from an OPC UA client.

AlinMoldovean, 2018-10-09:
@pepeevich ,

That can be possible in case of providing implementation for NodeManagement set of services in both client and server.
Another option is to expose some UA methods in the server (e.g. AddFolder, AddVariable)  to allow client to invoke them and create nodes.

pepeevich, 2018-10-12:
@AlinMoldovean Okay, thank you for the answer. The issue can be closed.

closed 2018-10-12 by AlinMoldovean