OPCFoundation/UA-.NETStandard #557: OPC UA Client Feature to Read Dynamic Types from the OPC UA Server

Noires, 2018-10-12:
As pointed out in #444 i finally managed to put together a rough sample to describe how to read dynamic types : https://github.com/Noires/OPCUAClientDynamicTypesSample . Since i wasn't able to reuse the code from the project, i had to redo it again. But it just shows the basic approach how it is done:

1. Find all custom enums and their encoded type id's on the server.
2. Find all custom types and their encoded type id's on the server.
3. Create a new assembly with a new module where the enums and types can be added.
4. Parse the description of the namespaces.
5. Create the enums for the parsed namespaces with their encoded type id's.
6. Create the types for the parsed namespaces with their encoded type id's and the created enums.
7. Add the created types to the encodeable factory.

There is room for a lot of improvement i think, but at this point i would rather prefer to implement it in the stack itself and make it a core feature. But for that i would need pointers where to implement which part of the feature.

MaartenBE, 2018-10-19:
This would be a great addition to the SDK in my opinion. I don't know enough about the internals of the stack to give you the required pointers, but I am certainly willing to help when it comes to testing.

MikeK-10, 2018-12-21:
Hi, I am struggling to get your example to work. I keep getting BadSessionIDInvalid exceptions when the code attempts to do _session.browse, sometimes the first call in FindEncodedEnumsIDs gives the exception, sometimes this is ok and FindEncodedTypeIds gives the exception but one of them always does. :(

MikeK-10, 2019-01-02:
Hi, is anyone able to assist with this? The FindEncodedTypeIds function returns an empty collection for foundcustomtypes ?

Noires, 2019-01-02:
Hi,
iam sorry for the late response, which was because of the holidays. I will try to look into it tomorrow.

MikeK-10, 2019-01-02:
Have made some progress now - my OPC UA server is Codesys and it appears that complex data types was not working in previous codesys version, with the new release of codesys foundcustomtypes now contains my custom types. I'm not at the point where everything is functioning but definitely made a leap forwards

MikeK-10, 2019-01-09:
> Hi,
> iam sorry for the late response, which was because of the holidays. I will try to look into it tomorrow and will hopefully have time to integrate it properly into the sdk soonish.

Hi, just wondered if you have been able to progress this at all? Seems like static creation of types is a dead end without .xml file but I am unable to get the dynamic type creation to work with types that contain other types.

MikeK-10, 2019-01-15:
Totally understand this isn't your day job and thanks for continuing to move forwards. 

The section before is the full type dictionary, I added Float to your select case statement, and the lowest level of my custom types (Application.Gripper) seems to function correctly (though I had to change the For loop to work backwards as Application.Gripper is the last entry). The problem is that a type using the gripper type e.g. Application.Plate gives an error when I create an instance of it and try to adjust properties.

Noires, 2019-01-15:
Okay, iam looking into it. Currently implemented your type dictionary in my own server and added some test data. I will check tomorrow what needs to be changed to properly adjust the properties there.

Noires, 2019-01-16:
I found some issues with the order of the type dictionary of yours, which is not resolved in the example i provided. In the example i just look from top to bottom in the type dictionary and create type by type. So he tries to create the array property for the Application.Plate with the Application.Gripper, but the type does not exist at this point. I will try to provide the changes to the example soonish.

MikeK-10, 2019-01-16:
Yes, I had this issue and resolved by reversing the order in which they are processed (change to the For loop), it was at this point I began to have the current issue.

Noires, 2019-01-16:
Ah, could you share the code to show how you try to change the properties and create the Application.Plate? I think i found the issue which was caused by a static property in the dynamic generation, which does not allow to change the properties. Just need to make sure that this is the issue.

MikeK-10, 2019-01-16:
Sent you via email - let me know if you do not receive

Noires, 2019-01-16:
Received the email, but i cant find the part where you try to change some properties? I can create a instance of Application.Plate and adjust the Application.Grippers after removing the Field.Attributes.Static as pointed above.

MikeK-10, 2019-01-16:
I didn't try to change any properties, just the direction of the For loop such that Application.Gripper is known when Application.Plate is created.  

Another issue is that on line
BuildDynamicProperty(typeBuilder, NodeIdString, typeof(uint), (uint)encodingId.Identifier);

encodingId.Identifier is equal to : "|type|CODESYS Control Win V3 x64.Application.Table"

so an exception occurs due to an Int being expected, I had to put a temporary integer in just to move on but this will cause problems later?

Noires, 2019-01-16:
Changing should be now possible with the change above. Any other help you need? Else i will work further on the first version of the feature.

MikeK-10, 2019-01-16:
Brilliant, hopefully this will allow me to progress for a while, like you I have other projects are currently demanding my time too so it may be a little while until I can say for sure. 

Will be excited to see how your work progresses, a fully flexible tool that can create all the custom types from server info then read all the variables from the server into an instance of the correct type would be a beautiful thing :)

MikeK-10, 2019-01-16:
Hi, I removed the static property, i.e. changed line to :
var field = typeBuilder.DefineField("_" + propertyName, propertyType, FieldAttributes.Private);

however this has made no difference, when I try to do anything with an instance of Application.plate I get the same error. 

[image]

Noires, 2019-01-16:
It might be because you are trying to adjust the property in Visual Studio and the Assembly is not known at that point. If that would be the case, then i have to look how you can make it known to VS.

MikeK-10, 2019-01-16:
Not sure, I can change fields of an instance of Application.Gripper but not Application.Plate or Application.Table, even the fields of Application.Table that should definitely be known types e.g. the Idle field.

Noires, 2019-01-16:
Strange, i will look into it again tomorrow. I think i had no issues with adjusting the properties for those types.

MikeK-10, 2019-01-21:
did you find anything?

Noires, 2019-01-24:
Sorry for the late response again. I currently have no issues with changing the properties of the instances. But i do not have the same configuration for the nodes like you i suppose. Are the fields which you try to change "mandatory" nodes? I can change fields like Idle without any issue.

Noires, 2019-01-24:
Maybe the issue is caused by the naming of your nodes. Basically i did not use the dot between Application.Plate and so on since it causes issues with the ModelCompiler. Can you change it for testing purpose?

MikeK-10, 2019-01-24:
I can't as this is what the codesys OPC UA server gives, I have no option to change this. I can change the name of Application or the name of the individual types but I cannot change the format which always has the dot present

MikeK-10, 2019-01-24:
Hi, I just tried this but it still gives the same issue:

[image]

Noires, 2019-01-24:
Hard to tell what is causing the issue since i cant reproduce it on my side and don't have your server for testing. You could just debug the type generation and check the fields if something seems off. At the moment i have no idea though why it might not work. Maybe it is even a Visual Studio Issue?

MikeK-10, 2019-01-24:
Everything seems fine so I don't know where to go. Codesys is a free download (https://store.codesys.com/codesys.html?___store=en) - which is why I am using this for initial testing - was planning to move to hardware PLCs next, but appreciate this is going to be more effort than you will probably want to put in.

Noires, 2019-01-24:
Are there many steps required to setup the nodes?

MikeK-10, 2019-01-24:
No, I will send you a codesys project in case you want to try

MikeK-10, 2019-02-05:
I found some instructions for OPC UA in codesys too:
https://factoryio.com/docs/tutorials/codesys/setting-up-codesys-opc-ua/

ga5tan, 2019-06-23:
@MikeK-10 how did you connect to CodeSys. Sample Client wont show CodeSys server in discovery :(

AlinMoldovean, 2020-01-16:
Complex types support for Client side implemented in #849

closed 2020-01-16 by AlinMoldovean