OPCFoundation/UA-.NETStandard #564: Subscriptions & Monitored items - max issues

RednasN, 2018-10-18:
Hi, 

I am doing some performance test with Subscriptions and Monitored items. I am using a Siemens PLC which is running a OPC UA Server (Medium license). In my test I want to find out the maximum monitored items I can subscribe to. I have found out that each subscription can hold a maximum of 1000 monitored items, if more it will throw a BadTooManyOperations exception. I think this is configured on the OPC Server. 

My solution to this is to create a new subscription when the maximum monitored items per subscription is reached.  This brings me to my next problem, when I have an array of 5 subscriptions with 976 monitored items each it will only "really" subscribe to the first 1-2 subscriptions when I execute ApplyChanges. When I check the MonitoredItemCount within the SubscriptionDiagnosticsArray it only shows 976 monitored items for the first two subscriptions, the others are empty.  

This shows the elapsed milliseconds for each subscription when I execute 'ApplyChanges':

`Apply Changes Sub Nr: 0- Monitored Items count: 976
Changes Sub Nr: 0-976 Elapsed ms: 18960
Apply Changes Sub Nr: 1- Monitored Items count: 976
Changes Sub Nr: 1-976 Elapsed ms: 3887
Apply Changes Sub Nr: 2- Monitored Items count: 976
Changes Sub Nr: 2-976 Elapsed ms: 553
Apply Changes Sub Nr: 3- Monitored Items count: 976
Changes Sub Nr: 3-976 Elapsed ms: 529
Apply Changes Sub Nr: 4- Monitored Items count: 976
Changes Sub Nr: 4-976 Elapsed ms: 543
Apply Changes Sub Nr: 5- Monitored Items count: 217
Changes Sub Nr: 5-217 Elapsed ms: 106`

Is this a known problem or I am doing something wrong?

This is my code I am testing with:

`Session session = _opcSessionManager.GetSession(opcConnectionName).Session;
            Node node = _opcDataAccessManager.GetNodeByIdentifier(tagName, opcConnectionName, true);

            //Create subscription if not exist or subscription count is larger than 975
            if (subscriptions.Count == 0 || subscriptions.Last().Sub.MonitoredItems.Count() > 975)
            {
                Subscription subscription = new Subscription(session.DefaultSubscription);

                subscription.PublishingEnabled = true;
                subscription.PublishingInterval = 2500;
                subscription.KeepAliveCount = 10;
                subscription.LifetimeCount = 10;
                subscription.MaxNotificationsPerPublish = 1000;
                subscription.Priority = 100;

                session.AddSubscription(subscription);
                subscription.Create();

                lock (subscriptions)
                {
                    subscriptions.Add(new SubHolder() { Sub = subscription, Changed = false });
                }
            }

            //Add monitored item.
            MonitoredItem monitoredItem = new MonitoredItem();

            monitoredItem.StartNodeId = node.NodeId;
            monitoredItem.AttributeId = Attributes.Value;
            monitoredItem.DisplayName = null;
            monitoredItem.MonitoringMode = MonitoringMode.Reporting;
            monitoredItem.SamplingInterval = 1000;
            monitoredItem.QueueSize = 0;
            monitoredItem.DiscardOldest = true;
            monitoredItem.Notification += MonitoredItem_Notification;

            lock (subscriptions)
            {
                subscriptions.Last().Sub.AddItem(monitoredItem);
                subscriptions.Last().Changed = true;
            }`

And, when all subscriptions are populated I execute this:
`for (int i = 0; i < subscriptions.Count; i++)
                {
                    if (subscriptions[i].Changed)
                    {
                        subscriptions[i].Sub.ApplyChanges();
                        subscriptions[i].Changed = false;                        
                    }
                }`

DummyRick, 2019-01-24:
Hi RednasN,

The number of 1000 items it's set by default on the OPC UA server.
If you wanna increase this number you should modify this settings from the server side.

AlinMoldovean, 2019-01-28:
Hi @RednasN ,

The limitation seems to be on the server side.
You should try the same scenario against one of the servers from this repository. e.g. ReferenceServer.

DummyRick, 2019-01-28:
@AlinMoldovean regarding the performance tests... As @RednasN but i'm trying to subscribe to10K nodes using a single session. 
Client takes more or less a minutes to be notified of all the nodes changes.

Reading the same amount of nodes using the normal read it's takes less than 1 second.

My questions are:
1. 1 minutes to read 10k nodes using subscriptions sound normal for you? Or it's should be much more fast?

2. if this amound of time is normal, Why i should use subscription istead of pooling?

RednasN, 2019-03-26:
Thanks for the replies, I found out that there are some limits within the OPC Server running on the Siemens PLC. 

For other people experiencing a problem like me, you can find more information up here:
[What are the system limits of the OPC UA Server with S7-1500 with firmware V2.0.x and V2.1.x?
](https://support.industry.siemens.com/cs/document/109755846/what-are-the-system-limits-of-the-opc-ua-server-with-s7-1500-with-firmware-v2-0-x-and-v2-1-x-?dti=0&lc=en-WW)

And how to increase timing/intervals:
[OPC UA .NET Client for the SIMATIC S7-1500 OPC UA Server](https://cache.industry.siemens.com/dl/files/901/109737901/att_898965/v3/109737901_OPC_UA_Client_S7-1500_DOKU_V10_en.pdf)

@DummyRick Reading (recursive) nodes takes a lot of time for me too. I can imagine reading 10k nodes in one minute is not really bad.

closed 2019-03-26 by RednasN

gustinmi, 2020-08-05:
Thanks for idea with subscription list. Beside that, there is also limit on session.browse (it fetches around 1000 items max). One should always use browsing in batches (browseNext) and continuation points. It work well with simatic.net opcua server component.