OPCFoundation/UA-.NETStandard #565: Connecting failed with error 'BadSecurityChecksFailed' with SDK Version 1.04.353.15

vijinpv, 2018-10-18:
I am doing an upgrade of OPCUA SDK from version v1.03.340 to v1.04.353.15. After the upgrade of SDK I am facing a connection issue from a client with Authentication. Connection works fine if it is anonymous. 
The same issue is observed with OPC UA expert tool as well. 
The ServiceResultException is raised inside the function RsaUtils.Decrypt() from server, which is called for UserNameIdentityToken decryption during the ActivateSession() method call. The exception is due to a missing private key in the server certificate. 
public static byte[] Decrypt(
            ArraySegment<byte> dataToDecrypt,
            X509Certificate2 encryptingCertificate,
            bool useOaep)
        {
            RSA rsa = null;
            try
            {
                rsa = encryptingCertificate.GetRSAPrivateKey();
                if (rsa == null)
                {
                    throw ServiceResultException.Create(StatusCodes.BadSecurityChecksFailed, "No private key for certificate.");
                } 
Our application uses a directory certificate store which is located at C:\ProgramData\<Organization\ folder. The server generates certificate only once during the start up. If you restart the server, it uses the certificate generated in previous session. 
The issue is not present if I clear all certificate in the location and start the server. During this time the certificates are generated and used. Issue will be reproducible only if you restart the server with a previous certificate.
What I see from my debugging is that the LoadPrivateKey() method in CertificateIdentifier is called only once when you generate a new certificate. For all other sessions it is not called. 
I have tried a workaround as of now for my testing. I have just tried to call LoadPrivateKey() method and pass the certificate while creating a session. This seems to solve the issue.

Please let me know if it is a known bug. Is there any fix for SDK?. If not what could be the issue with my server. 
Is not calling LoadPrivateKey() method on the next session expected?

vijinpv, 2018-10-18:
[image]

mregen, 2018-10-26:
@vijinpv Hi, could you share your app startup code? and config file? There might be some configuration issue which prevents the code from finding the .pfx on a subsequent start.

closed 2020-01-30 by mregen

jabire, 2023-12-12:
How this is get fixed

mregen, 2023-12-14:
If you don't want to use the file configuration, please use the app configuration with fluent API and don't fill out the structures directly, as you miss some internal sanity checking when the configuration is built which may lead to strange symptoms.