OPCFoundation/UA-.NETStandard #566: Durable Subscription

vegard77, 2018-10-19:
Hi,

I'm building an OPC UA ClientAndServer application in C# .NET.
I first built the code using SDK 1.02.336
I  have just spent some time now to update my code to support SDK version 1.04.353 - in order to be able to implement DurableSubscription .

I got a nice suggestion from a guy Jim At the OPC Foundation Forum and he advised me to implement DurableSubscription to support my needs. He gave me the following advise::

**In order for a Client to completely disconnect from a Server and later reconnect to a previous Subscription and retrieve the queued values**:

1. The Server must support the feature “durable Subscriptions” (not many Servers support this feature !!!)
2. The Client must explicitly set the Subscription durable by calling the SetSubscriptionDurable Method.

See Clause 6.8 of OPC UA Part 4 http://www.opcfoundation.org/UA/Part4/ for more details to use this feature.

This is exactly what I want to achieve!
I have been reading the documentation in part 4 and 5, but struggle to get any further. 
I cannot find a single example anywhere on how to implement this feature.  Can anyone give me some advice or even better post some code that gives me a hint.  In the SDK I find the class SetSubscriptionDurableMethodState. I guess I have to do some implementation on the severside to let the client set the subscription in Durable mode. 

Thanks Vegard

AlinMoldovean, 2018-10-19:
Hello @vegard77 ,

The required steps on the server side are:

- Implement logic for _SetSubscriptionDurable_ method and set the subscription in a _durable_ state where it won't be deleted at CloseSession or lifetime expiration

- Implement _TransferSubscriptions_ service 

That will allow the clients to use this "durable subscriptions" feature.

vegard77, 2018-10-29:
Hi,

I struggle to implement the TransferSubscription service.
Described in the OPC Specs Part 4 (Services 1.04)

On my client I call the TransferSubscription:

`TransferSubscriptionsRequest req1 = new TransferSubscriptionsRequest();
req1.SubscriptionIds = subs;
req1.SendInitialValues = true;
ress = m_session.TransferSubscriptions(req1.RequestHeader, subs, true, out transfer, out diagnostic);`

On the Serverside I struggle to find the suitable methods to use, described in the OPC Specs Part 4 like:

- SetSession()

Set the Session information for the Subscription to match the Session on which the TransferSubscriptions request was issued.

- SessionChanged()

A boolean function that returns TRUE only when the Session used to send a TransferSubscriptions request is different from the Client Session currently associated with the Subscription.

How to I assign the new sessionId to subscriptions that match the collection of SubscriptionsIds to Transfer. Like SetSession() described above. I cannot find these methods at all???

`public override ResponseHeader TransferSubscriptions(
            RequestHeader requestHeader,
            UInt32Collection subscriptionIds,
            bool sendInitialValues,
            out TransferResultCollection results,
            out DiagnosticInfoCollection diagnosticInfos)
        {
            results = null;
            diagnosticInfos = null;

            ValidateRequest(requestHeader);
            
            lock (Lock)
            {
                
                TransferResultCollection transfer = new TransferResultCollection();
                
                IList<Subscription> subList = serverObject.SubscriptionManager.GetSubscriptions();
                foreach(Subscription sub in subList)
                {                
                    foreach (Session sess in serverObject.SessionManager.GetSessions())
                    {
                        if(subscriptionIds.Contains(sub.Id))// && (sub.Session==null ))
                        {
                            //sub.setSession == sess;
                        }
                        //if(s.ClientCertificate.==requestHeader.AuthenticationToken.Identifier)
                        //foreach(Subscription sub in serverObject.SubscriptionManager.GetSubscriptions())
                        //{
                        //    if(sub.Session==null)
                        //    {
                        //        sub.SessionId==
                        //    }
                        //    else if(sub.Session.IdentityToken.==requestHeader.AuthenticationToken.Identifier)
                        //}
                    }
                }
            }
            // Insert implementation.

            //return CreateResponse(requestHeader, StatusCodes.BadServiceUnsupported);
            return CreateResponse(requestHeader, StatusCodes.GoodSubscriptionTransferred);
        }
`
Please give some hints or even better an example implementation...

Regards Vegard

closed 2019-09-26 by AlinMoldovean