OPCFoundation/UA-.NETStandard #573: OPCUA Subscription number of reported items

MrLonko, 2018-10-26:
Hi experts!

It is not an issue but more a question that i cannot wrap my head around. 
If we subscribe to OPCUA Server with lets say 100 nodes and we register to events:
```
 eachItem.Notification += OnNotification;
 subscription.AddItems(list);
```

Now for example next time server publishes 20 nodes changes, we get a call for each on OnNotification callback. If we want to agregate data i would like to wait for all 20 nodes to report. Is there a counter to know how much items are being reported at one time ? Or can sync read them all in a loop and have one master notification?

So is there a callback notification saying you have new data comming from server in which we check how many nodes are in this interval beeing published and read the in a loop ?

Sorry if the question is hectic but, we are still learning the whole stack.

With kind regards,

AlinMoldovean, 2018-10-26:
Hi @MrLonko ,

You can use Subscription.FastDataChangeCallback event handler.
It will provide the complete publish response in a single notification.
Something like this:

```
m_subscription.FastDataChangeCallback = OnDataChange;

 private void OnDataChange(Subscription subscription, DataChangeNotification notification, IList<string> stringTable)
        {
            try
            {
                foreach (MonitoredItemNotification itemNotification in notification.MonitoredItems)
                {
                    MonitoredItem monitoredItem = subscription.FindItemByClientHandle(itemNotification.ClientHandle);

                    if (monitoredItem == null)
                    {
                        continue;
                    }
                     // itemNotification.Value.Value contains the data change of the monitored item.

                }
            }
            catch (Exception exception)
            {
                
            }
        }
```

MrLonko, 2018-10-26:
Wow excelent.... AND FAST!!!  THANK YOU

Just a quick question regarding this line:
subscription.FindItemByClientHandle(itemNotification.ClientHandle);

Why do i need to find item by clienthandle ? Arent all notified items from my connection ? eg. My client?

BR

AlinMoldovean, 2018-10-26:
You need FindItemByClientHandle only only if you want the reference to corresponding MonitoredItem object.
Otherwise you don't need it..

MrLonko, 2018-10-26:
If i just need value and node id I have no need to call it. Excelent!

Thank you very much!

closed 2018-11-29 by AlinMoldovean