OPCFoundation/UA-.NETStandard #579: Using Username / Password Authentication "help wanted"

ciappara, 2018-10-30:
Hello everyone,

This is not an issue but a question about authentication as I cannot wrap my head around it. I posted this issue in StackOverflow (no replies yet) and I thought that here might be the best place to ask.

I’m trying to authenticate to an OPC UA Simulation Server using the username/password authentication but I’m receiving the below ***Bad_SecurityChecksFailed*** exception (most probably due to a certificate error). It's the first time I'm using OPC UA, and I have no prior experience in using it. Maybe someone already encountered this same issue, and can help me out.

[![enter image description here][1]][1]
 
On the simulation server that I’m using (Prosys OPC UA), it’s showing that the certificate was rejected:

[![enter image description here][2]][2]

This is the *UserIdentity* variable’s data:

[![enter image description here][3]][3]

As can be seen, the IssuedTokenType and PolicyId are null which might be part of the issue, but I cannot find how to initialise them.

Here is the *ConfiguredEndpoint* data:

[![enter image description here][4]][4] 

Below is the code to create a session on the server using the username and password:

[![enter image description here][5]][5]

When creating the Session, it throws an AggregateException with the above ***Bad_SecurityChecksFailed*** exception.

And these are the endpoints and users setup on the simulation server:

[![enter image description here][6]][6]

[![enter image description here][7]][7] 

I searched for C# examples but I didn't manage to find what I was looking for.

I also checked the 'UA Quickstart applications' project which if I understood correctly, it's using certificates when using the username and password authentication. I don't have any experience with certificates and I didn't manage to understand how the authentication is working, and how I can implement it.

Thank you in advance,
Clive

  [1]: https://i.stack.imgur.com/rh3wO.png
  [2]: https://i.stack.imgur.com/2A9TC.png
  [3]: https://i.stack.imgur.com/94xle.png
  [4]: https://i.stack.imgur.com/VG0aR.png
  [5]: https://i.stack.imgur.com/REUlw.png
  [6]: https://i.stack.imgur.com/v1Af1.png
  [7]: https://i.stack.imgur.com/NGbtf.png

awcullen, 2018-10-30:
Hi Clive,
I see in the server log that the server rejected the remote certificate (the certificate coming from the client).  Most servers reject any new client certificate, by default. Prosys Simulation Server asks you to click on the Certificates tab and find the new client's certificate, then right-click on it and select 'trust'.

ciappara, 2018-11-02:
Hi Andrew,

Thank you for your reply. I did as you said, trusting the certificate manually once created but it was still giving the exception, since the "Session.Create" part will fail if the certificate isn't already trusted.

Then I used Polly to wait and retry the session creation. The first time it's created, it passes the untrusted certificate which shows up on the simulation server. Now I can trust the new certificate on the server, and the next time the "Session.Create" tried to recreate a session with the same certificate, it was successful.

[image]

Not sure if there's a better way to deal with the untrusted certificate, or maybe how to send a trusted certificate to the OPC UA server from the get-go.

Thank you for your help!
Clive

PaulHunkar, 2018-11-02:
Clive,

The certificates are used for security - which does require some setup.  The idea is that only a trusted client can connect to the server.  In a real world deployment the certificates are usually issued by a Certificate Authority and then all certificates issued by the CA are trusted.  The client should only connect the servers that it trusts and server only accept clients that are trusted.    All application save certificate in a certificate store (defined by the application) and a user with administrative rights can access the store and load the public key part of a certificate into the store.  i.e. the trust list can be pre-configured.

closed 2018-12-13 by AlinMoldovean