OPCFoundation/UA-.NETStandard #585: Close connection with clients by server

BadIrbis, 2018-11-08:
Testing this OPC UA Server with a third-party OPC UA client.
I can not understand how to break the connection with the client on the initiative of the server. The designed program provides the ability to start and stop the server. For this, I use `Application.Start()` to run and `Application.Stop()` to stop, as was done in the quickstart examples.
The problem is that after stopping the server, the client is still connected to it. This can be seen in the server logs and the display of the connection status to the server in the client. If after that you close the application from which the server was started, the client displays the loss of connection. If you do not close the application, but disable the client, you can see the disconnection in server logs.
So, is it possible to disconnect clients on server stop?

AlinMoldovean, 2018-11-08:
Hi @BadIrbis ,

I made a test with Server.Stop() methods an it is behaving as expected, i.e. it closes the existing client connections.
The proper way to destroy a server object is:
```
 Server.Stop();
 Server.Dispose();
 Server = null;
```

BadIrbis, 2018-11-08:
Thanks for reply @AlinMoldovean . Have no idea what can be wrong, but it doesn't work for me.
Here is what I did for testing this time - i took Quickstarts.EmptyServer and added "Stop" button to menu.

```
        private void stopToolStripMenuItem_Click(object sender, EventArgs e)
        {
            try
            {
                m_server.Stop();
                m_server.Dispose();
                m_server = null;
            }
            catch (Exception exception)
            {
                Utils.Trace(exception, "Error stopping server.");
            }
        }
```

Then I turned on this server, connected to it through Quickstarts.EmptyClient and stopped server. Quickstarts client reported disconnect, seems some trouble in client I used first. But in server logs we still can see some requests and responces.

Server start:
```
08.11.2018 14:47:41.210 Creating the Node Managers.
08.11.2018 14:47:42.113 Server: Session Monitor Thread Started.
08.11.2018 14:47:42.118 Server: Publish Subscriptions Thread Started.
08.11.2018 14:47:42.685 Channel 1 in Connecting state.
08.11.2018 14:47:42.695 TCPSERVERCHANNEL SOCKET ATTACHED: 005C39D4, ChannelId=1
08.11.2018 14:47:42.706 Channel 2 in Connecting state.
08.11.2018 14:47:42.715 Channel 1 in Opening state.
08.11.2018 14:47:42.715 TCPSERVERCHANNEL SOCKET ATTACHED: 01400C58, ChannelId=2
08.11.2018 14:47:42.764 TCPSERVERCHANNEL ForceChannelFault Socket=01400C58, ChannelId=0, TokenId=0, Reason=Id: BadConnectionClosed
Description: Remote side closed connection
08.11.2018 14:47:42.771 Channel 2 in Faulted state.
08.11.2018 14:47:42.828 Security Policy: http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
08.11.2018 14:47:42.836 Sender Certificate: CN=Quickstart Empty Client, C=US, S=Arizona, O=SomeCompany, DC=morv104
08.11.2018 14:47:42.843 Token #0 created. CreatedAt = 11:47:42.842 . Lifetime = 3600000
08.11.2018 14:47:42.859 SECURE CHANNEL CREATED [.NetStandard ServerChannel UA-TCP 1.4.354.0] [ID=1] Connected To: opc.tcp://morv104:62546/Quickstarts/EmptyServer [SignAndEncrypt/Basic256Sha256/Binary] Client Certificate: [CN=Quickstart Empty Client, C=US, S=Arizona, O=SomeCompany, DC=morv104] [798F35B05389EEBFBF997D4F1135D0E0FDA51F44] Server Certificate: [CN=Quickstart Empty Server, C=US, S=Arizona, O=SomeCompany, DC=morv104] [72F165EECBE57C7D0FF834945AFE58B3F7C895AA]
08.11.2018 14:47:42.869 Token #1 activated. CreatedAt = 11:47:42.842 . Lifetime = 3600000
08.11.2018 14:47:42.875 Channel 1 in Open state.
08.11.2018 14:47:42.882 Channel 1: SendOpenSecureChannelResponse()
08.11.2018 14:47:42.989 Channel 1: ProcessRequestMessage 1
08.11.2018 14:47:43.052 Session CREATED, Id=ns=3;i=2101607106, Name=Quickstart Empty Client, ChannelId=0110c77d-63ac-4335-8949-6b40b5806b40-1, User=Anonymous
08.11.2018 14:47:43.071 Server - SESSION CREATED. SessionId=ns=3;i=2101607106
08.11.2018 14:47:43.083 Channel 1: SendResponse 1
08.11.2018 14:47:43.118 Channel 1: ProcessRequestMessage 3
08.11.2018 14:47:43.133 Session VALIDATED, Id=ns=3;i=2101607106, Name=Quickstart Empty Client, ChannelId=0110c77d-63ac-4335-8949-6b40b5806b40-1, User=Anonymous
08.11.2018 14:47:43.141 Session FIRST ACTIVATION, Id=ns=3;i=2101607106, Name=Quickstart Empty Client, ChannelId=0110c77d-63ac-4335-8949-6b40b5806b40-1, User=Anonymous
08.11.2018 14:47:43.151 Server - SESSION ACTIVATED.
08.11.2018 14:47:43.159 Channel 1: SendResponse 3
08.11.2018 14:47:43.197 Channel 1: ProcessRequestMessage 4
08.11.2018 14:47:43.226 Channel 1: SendResponse 4
08.11.2018 14:47:43.232 Channel 0 in Connecting state.
...

```

Server stop:
```
08.11.2018 14:48:19.886 Server: Publish Subscriptions Thread Exited Normally.
08.11.2018 14:48:19.887 Server: Session Monitor Thread Exited Normally.
An exception was thrown: "Opc.Ua.ServiceResultException" at Opc.Ua.Server.dll
An exception was thrown: "Opc.Ua.ServiceResultException" at Opc.Ua.Server.dll
08.11.2018 14:48:23.266 Channel 1: ProcessRequestMessage 39
08.11.2018 14:48:23.281 Channel 1: SendResponse 39
An exception was thrown: "Opc.Ua.ServiceResultException" at Opc.Ua.Server.dll
An exception was thrown: "Opc.Ua.ServiceResultException" at Opc.Ua.Server.dll
08.11.2018 14:48:28.304 Channel 1: ProcessRequestMessage 40
08.11.2018 14:48:28.314 Channel 1: SendResponse 40
An exception was thrown: "Opc.Ua.ServiceResultException" at Opc.Ua.Server.dll
An exception was thrown: "Opc.Ua.ServiceResultException" at Opc.Ua.Server.dll
08.11.2018 14:48:33.284 Channel 1: ProcessRequestMessage 41
08.11.2018 14:48:33.370 Channel 1: SendResponse 41
An exception was thrown: "Opc.Ua.ServiceResultException" at Opc.Ua.Server.dll
An exception was thrown: "Opc.Ua.ServiceResultException" at Opc.Ua.Server.dll
08.11.2018 14:48:38.294 Channel 1: ProcessRequestMessage 42
08.11.2018 14:48:38.407 Channel 1: SendResponse 42
```

Exceptions are BadServerHalted:
[image]

But why there are some ProcessRequestMessage and SendResponse after server stop? Do we still have some sockets open after stop? With Quickstart client there are only few messages after stop, but with that third-party client messages do not stop until manual client shutdown.

BadIrbis, 2018-11-09:
So, yes, connected socket still alive. Screenshot with 2 netstat results - before and after stop. 
Before stop local machine listening out port and new clients can connect. Address rfc1918 is a virtual box VM with OPC UA client, which established connection to server.
After stop localhost stops listening port, new clients can't connect, but connection to client still established. It will close on client application close, or server application close.

[image]

I tryed some very dirty solution to test. Created public static list of sockets at `Opc.Ua.Bindings.TcpServerChannel`. Each socket, used in `Opc.Ua.Bindings.TcpServerChannel.Attach` added to that list. In my .NETFramework application, before stop, I manualy closing all sockets. I understand it's not correct, but now it works - no more esablished connection after stop.

```
            foreach (System.Net.Sockets.Socket s in TcpServerChannel.staticSocketList)
            {
                try
                {
                    s.Disconnect(false);
                }
                catch { }
            }
```

I do not have experience with .NETStandart, so maybe someone knows how to do it correct. As I understand, we close sockets at `Opc.Ua.Bindings.UaTcpChannelListener.Stop()`:

```
public void Stop()
        {
            lock (m_lock)
            {
                if (m_listeningSocket != null)
                {
                    m_listeningSocket.Dispose();
                    m_listeningSocket = null;
                }

                if (m_listeningSocketIPv6 != null)
                {
                    m_listeningSocketIPv6.Dispose();
                    m_listeningSocketIPv6 = null;
                }
            }
        }
```

And we can't use `m_listeningSocket.Disconnect(false)` here, because `Opc.Ua.Core(netstandart1.3).Socket` does not contains defenition for `Disconnect()`

closed 2020-01-16 by AlinMoldovean