OPCFoundation/UA-.NETStandard #586: OPC UA Client MonitoredItem GetServiceResult() might crash

cacamille3, 2018-11-10:
I think there is a problem in _SampleApplications/SDK/Opc.Ua.Client/MonitoredItem.cs_

In both [` GetServiceResult(IEncodeable notification, int index) `](https://github.com/OPCFoundation/UA-.NETStandard/blob/0d08d4018012bf0fb5ee5e535a0841205fcad370/SampleApplications/SDK/Opc.Ua.Client/MonitoredItem.cs#L943) and [` GetServiceResult(IEncodeable notification) `](https://github.com/OPCFoundation/UA-.NETStandard/blob/0d08d4018012bf0fb5ee5e535a0841205fcad370/SampleApplications/SDK/Opc.Ua.Client/MonitoredItem.cs#L921) notification message are assigned but wrongly checked against `!= null`

This might result in returning null for good Data MonitoredItem or Event Notification and also crash when trying to access `NotificationMessage`'s `StringTable` due to null dereference.

AlinMoldovean, 2018-11-13:
@cacamille3 ,

It seems to be a wrong null check indeed.
Could you provide a pull request with a fix for this?

cacamille3, 2018-11-14:
Added #589 Fixed GetServiceResult null checks

AlinMoldovean, 2018-11-14:
Fixed with commit 3546e56e05837c3a070ce2d8388c10789f6cb819

closed 2018-11-14 by AlinMoldovean