OPCFoundation/UA-.NETStandard #591: Could not load HTTPS Stack Listener.

ggokka, 2018-11-17:
step1. create windows service project in the Visual Studio
step2. add nuget library https://www.nuget.org/packages/OPCFoundation.NetStandard.Opc.Ua/
step3. update latetest version of all the nuget library
step4. write my own opc ua server 

Finally I got the error log when I launch the my own opc server

2018-11-17 19:10:42.200 Checking application instance certificate.
2018-11-17 19:10:42.235 Checking application instance certificate. CN=OPC Edge Pro Server, C=KR, S=Gyeonggi, O=UVC, DC=desktop-5b4ogmf
2018-11-17 19:10:42.953 Checking domains in certificate. CN=OPC Edge Pro Server, C=KR, S=Gyeonggi, O=UVC, DC=desktop-5b4ogmf
2018-11-17 19:10:43.743 The server is starting.
2018-11-17 19:10:44.132 Could not load HTTPS Stack Listener. TypeLoadException ''Microsoft.AspNetCore.Server.Kestrel.Https, Version=2.1.3.0, Culture=neutral, PublicKeyToken=adb9793829ddae60' 'Microsoft.AspNetCore.Server.Kestrel.Https.HttpsConnectionFilterOptions' Type Could not load

========================================
Id: Bad
Description: 'Microsoft.AspNetCore.Server.Kestrel.Https, Version=2.1.3.0, Culture=neutral, PublicKeyToken=adb9793829ddae60' 'Microsoft.AspNetCore.Server.Kestrel.Https.HttpsConnectionFilterOptions' Type Could not load
>>> 'Microsoft.AspNetCore.Server.Kestrel.Https, Version=2.1.3.0, Culture=neutral, PublicKeyToken=adb9793829ddae60' 어셈블리에서 'Microsoft.AspNetCore.Server.Kestrel.Https.HttpsConnectionFilterOptions' Type Could not load
---    위치: Opc.Ua.Bindings.UaHttpsChannelListener.Start()
---    위치: Opc.Ua.Bindings.UaHttpsChannelListener.Open(Uri baseAddress, TransportListenerSettings settings, ITransportListenerCallback callback)
---    위치: Opc.Ua.ServerBase.CreateHttpsServiceHost(IDictionary`2 hosts, ApplicationConfiguration configuration, IList`1 baseAddresses, ApplicationDescription serverDescription, List`1 securityPolicies)
========================================

Tratcher, 2018-11-26:
See https://github.com/aspnet/Announcements/issues/329. Your project will need binding redirects to work properly. Nuget should have added those for you.

closed 2019-02-21 by FischerSeb

FischerSeb, 2019-02-21:
See last comment.