OPCFoundation/UA-.NETStandard #596: How build the Opc.Ua.PredefinedNodes.uanodes?

codekimchi, 2018-11-24:
I found Opc.Ua.PredefinedNodes.uanodes in the Opc.Ua.Core project.

However  I don't know how could I rebuild the file.

AlinMoldovean, 2018-11-26:
@codekimchi ,

The PredefinedNodes.uanodes file is imported from [UA-Nodeset](https://github.com/OPCFoundation/UA-Nodeset/tree/master/Schema) repository.
It is built using [UA_ModelCompiler](https://github.com/OPCFoundation/UA-ModelCompiler) tool.

codekimchi, 2018-11-27:
@AlinMoldovean I plan to modify ModelDesign.xml file to add ModellingRule nodes. I need some ModelDesign Reference, Guide, Wiki etc.

cpipero, 2018-11-27:
If you are just trying to modify a modelDesign file you might want to try https://github.com/beeond/opcua-modeling-tool

--cos

ggokka, 2018-11-27:
@cpipero  Do you have a binary installer?

cpipero, 2018-11-27:
We maintain a free binary at https://beeond.net/opc-ua-products/

mpostol, 2018-11-27:
@codekimchi The documentation that may be useful for you is here:

[OPC UA Information Model Deployment](http://www.commsvr.com/DownloadCenter/Publications/OPCUAInformationModelDeployment.aspx)

The designer integrated with the help documentation is available here:
[OPC UA Address Space Model Designer](http://www.commsvr.com/Products/OPCUA/UAModelDesigner.aspx)

Hope it helps.

codekimchi, 2018-11-27:
@cpipero I tried to open the belew file
https://github.com/OPCFoundation/UA-ModelCompiler/blob/master/ModelCompiler/Design/StandardTypes.xml

However I got this error
[image]

Why?

cpipero, 2018-11-30:
@codekimchi ,

I replicated the problem and it happens on my PC as well.

I submitted an issue to that repo https://github.com/beeond/opcua-modeling-tool/issues/5 .
In the meanwhile, you can open the models from the C:\Program Files (x86)\Beeond\UMX\samples folder (those are standard models from different OPC Foundation working groups)

https://github.com/OPCFoundation/UA-ModelCompiler/blob/master/ModelCompiler/Design/StandardTypes.xml

[image]<https://user-images.githubusercontent.com/40758405/49099875-f0ef0a00-f2b5-11e8-9d19-afd74aa28b2c.png>

closed 2018-12-13 by AlinMoldovean