OPCFoundation/UA-.NETStandard #597: OPCUA client resilience

MrLonko, 2018-11-26:
Dear,

We create a subscription and server is reporting data back. Once for instance there is network disconnection we start receiving reconnect notification. What and how to handle once reconnecting has come to a stop ?
How to know when keepalive are not sent anymore and we should start the communication from scratch.
How to kill existing connection ?

With kind regards,

MrLonko, 2018-11-29:
@AlinMoldovean first of all thank you, will look at handler.

we have an OPCUA server that supports only 2 sessions. We are doing resillence testing. What we are noticing that when we pull the network cable, from 4-10 minutes we are getting BADREQUEST from client stack. Which is similar to server not allowing another connection.

Is there a way to set session timeout from client to server, to lower this time ? I see some combination of keepalives etc... but don't really understand how they affect when opcua server (unified automation) knows when to remove old session and allow new.

AlinMoldovean, 2018-11-29:
@MrLonko ,

You need to request a lower value for _sessionTimeout_ parameter when creating the session with Session.Create() . e.g. 5000(ms).
```
public static async Task<Session> Create(
            ApplicationConfiguration configuration,
            ConfiguredEndpoint endpoint,
            bool updateBeforeConnect,
            bool checkDomain,
            string sessionName,
            uint **sessionTimeout**,
            IUserIdentity identity,
            IList<string> preferredLocales)
```

closed 2019-09-26 by AlinMoldovean