OPCFoundation/UA-.NETStandard #600: Create nuget packge without HTTPS support

K47S, 2018-11-27:
Generating a nuget package without HTTPS Endpoint support would be appreciated because it would remove the dependency to:
https://www.nuget.org/packages/Microsoft.AspNetCore.Server.Kestrel
and 
https://www.nuget.org/packages/Microsoft.AspNetCore.Server.Kestrel.Https
and therefor slim down the dlls count and size enormously.

K47S, 2018-12-03:
[Pull request with nuspec for NoHttps](https://github.com/OPCFoundation/UA-.NETStandard/pull/605)

AlinMoldovean, 2018-12-13:
PR was rejected since we do not plan to release a separate nuget for this.

mregen, 2020-10-14:
This topic is back on deck because we can release separate packages now...

closed 2020-10-20 by mregen

mregen, 2020-10-21:
see 1.4.364.28-preview packages

commit 62eedc19 (2020-10-20): Reduce core dependencies, move https binding to optional assembly. (#1157)

* implement transport binding factory
* split https transport into seperate nuget package
* new Opc.Ua.Binding.Https assembly
* Https binding is loaded on demand, if available
* reduce core dependencies to kestrel and others
fixes #600