OPCFoundation/UA-.NETStandard #602: Matlab opc toolbox

frheuinghhn, 2018-11-28:
Hey guys, 
I am working on the communication between an opc ua server (with python) and a client in matlab. Im trying to read the data from the server and to display them as a graph in matlab. My problem is that I can't find any example how to read value from a node with opcua. I have read the descriptions in mathworks already, but I couldnt find out how to get a list of all the nodes on the server and their value for example.
Hope somebody can help me.

AlinMoldovean, 2018-11-28:
@frheuinghhn ,

In order to get the list of all nodes of the server you need to make a recursive Browse on the server's address space, and for that you should use the Client.Session.Browse() method.
For reading you can use the Client.Session.ReadValue(NodeId nodeId) method.

frheuinghhn, 2018-11-28:
@AlinMoldovean ,
thanks for the answer. Unfortunately I couldnt find any of these methods in matlab.

AlinMoldovean, 2018-11-29:
@frheuinghhn ,

This repository hosts the reference implementation for the OPC UA Stack of OPC Foundation based on .NET Standard.
Do you use this library for your OPC UA Client implementation?

frheuinghhn, 2018-11-29:
@AlinMoldovean I am implementing my Server on python (pydev) and my client on Matlab. I couldnt find any of these methods on matlab. 
However I came a bit further with my programm. Im now able to read some of the values with matlab. The only problem I have now is that I cant read a whole array of data from server. 
This is how my code looks like: 
Python: 

radar = objects.add_object(idx, "Radar", radar_type.nodeid)
myvar = radar.add_variable(0, "Radardaten", ua.Variant([], ua.VariantType.Float))

server.start()
    # enable data change history for this particular node, must be called after start since it uses subscription
    server.historize_node_data_change(myvar, period=None)

    try:
        radarboard = TiRadarFile(showPlot=False)
        while True:
            nextFrame = radarboard.getNextFrame() 
            if nextFrame is None:
                break
            a = nextFrame.tolist()
            myvar.set_value(a)                                                                                                                    
            if nextFrame is None:
                break
                    
    finally:
        # close connection, remove subscriptions, etc
        server.stop()

myvar is an array of float. I am trying to display the values of this array on matlab

in Matlab 
Daten  = findNodeByName(Radar, 'Radardaten');
[val,time,qual] = readValue(client,Daten);
disp(val)
disp(time)
disp(qual)

all I get is an empty array.

fchaxel, 2018-11-29:
Hi @AlinMoldovean 

As I understand reading Fig 9, Part 3 Address Space model, a recursive browse call cannot be done without care since cyclic paths are allows.

I have already implemented such loop in a server address space.

Bye.

closed 2018-12-13 by AlinMoldovean