OPCFoundation/UA-.NETStandard #616: Recording time

MintLavender, 2018-12-17:
Hello.
Sorry for my English.
I'm interested in the rate at which node values are written. When recording 100 nodes, the average time is 25 seconds. How can this time be improved?
Here is the code I use:

> private string Write(WriteValueCollection nodesToWrite)
        {
            RequestHeader requestHeader = new RequestHeader();
            requestHeader.ReturnDiagnostics = (uint)DiagnosticsMasks.None;
            StatusCodeCollection results = null;
            DiagnosticInfoCollection diagnosticInfos = null;
            ResponseHeader responseHeader = null;
            try
            {
                responseHeader = session.Write(
                requestHeader,
                nodesToWrite,
                out results,
                out diagnosticInfos);
            }
            catch (Exception ex)
            {
                return "Error writing values: " + ex.Message;
            }
            ClientBase.ValidateResponse(results, nodesToWrite);
            ClientBase.ValidateDiagnosticInfos(diagnosticInfos, nodesToWrite);
            if (StatusCode.IsBad(results[0]))
            {
                return "Error: Sataus code" + results[0];
            }
            return "";
        }

AlinMoldovean, 2018-12-17:
Hello @MintLavender ,

A delay of 25 seconds for writing 100 values in a single Write request is quite high.
The code you are using on the client side is ok. 
The delay seems to be on the server side and that is the place to further investigate.

MintLavender, 2018-12-17:
Thanks for the answer @AlinMoldovean . I'll deal with the server.

MintLavender, 2018-12-18:
Thanks for the help! You were right the problem was due to the server. Something interferes with the server, but everything works fine on a virtual machine.

closed 2018-12-18 by MintLavender