OPCFoundation/UA-.NETStandard #617: Where do I count monitored items?

ggokka, 2018-12-19:
I found a method 'CreateMonitoredItems' in the CustomNodeManager.cs file. However unfortunately it doesn't work when new monitored item is created. 

Would I get another methods? 
I just want to limit monitored items due to the performance issues.

RednasN, 2018-12-20:
I am not sure if I understood you correctly, but you can view the MonitoredItemCount within your subscription (Subscription.MonitoredItems.Count). You can also view the statistics of the server , something like: Server >> ServerDiagnostics >> SubscriptionDiagnosticsArray >> your subscription >> MonitoredItemCount.

AlinMoldovean, 2019-01-21:
@ggokka ,

You can create an override of the StandardServer.CreateMonitoredItems() method and check the total existing monitored items before calling the base implementation.

NakWarsi, 2019-02-08:
@ggokka  my requirement was same as you have mentioned
i tried exactly what @AlinMoldovean suggested,
and it's working fine no i am able to make the monitored item limit.
thanks to @AlinMoldovean

closed 2019-02-13 by AlinMoldovean