OPCFoundation/UA-.NETStandard #623: Opc.Ua.ServiceResultException:“Could not load configuration file.”

Federau, 2019-01-05:
Hi:
   'when I built a new dot not core console project . I  met  the problem  as is written in title.  
  my program is shown below:
  class Program
    {
        static void  Main(string[] args)
        {
            Test();
            Console.Read();
        }
        static async void Test() {
            ApplicationInstance instance = new ApplicationInstance()
            {
                ApplicationName = "UA Core Sample Client",
                ApplicationType = ApplicationType.Client,
                ConfigSectionName = Utils.IsRunningOnMono() ? "Opc.Ua.MonoSampleClient" : "Opc.Ua.SampleClient"
            };
            var configuration = await instance.LoadApplicationConfiguration(false);
        }
    }

Federau, 2019-01-05:
the exception is thrown in line "var configuration = await instance.LoadApplicationConfiguration(false);"

AlinMoldovean, 2019-01-14:
@Federau, 
Please try using the _public async Task<ApplicationConfiguration> LoadApplicationConfiguration(string filePath, bool silent)_ method instead of _LoadApplicationConfiguration(bool silent)_

NakWarsi, 2019-01-23:
@Federau 

it seems like everything in your code is absolutely alright,
just check if you have missed import opcua configuration namespace "using Opc.Ua.Configuration;"
and try giving the variable type explicitly like this,

`  ApplicationConfiguration config = await application.LoadApplicationConfiguration(false);`

and if it's not working even then there must be the have some error occured in config file by mistake,
or any read error,
so finlly try providing external configuration file(xyz.Config.xml);

Reference Code:-
`            ApplicationInstance application = new ApplicationInstance
            {
                ApplicationName = "ClientTest2",
                ApplicationType = ApplicationType.Server,
                ConfigSectionName = "xyz"
            };

            // load the application configuration.
            ApplicationConfiguration config = await application.LoadApplicationConfiguration(false);`

closed 2019-05-23 by AlinMoldovean

inmen, 2020-05-14:
need to add lib 'System.ServiceModel.Primitives'
in Opc.Ua.Configuration need it for serialize

Havlat82, 2020-06-30:
Hi, I found myself stuck in exactly the same problem. Have you checked your .Config.xml properties and set "Copy To Output Directory" to "Copy if newer" in option? That caused exception in my case.