OPCFoundation/UA-.NETStandard #624: Trying to Configure MQTT DataWriter, fail to configure Publisher DataSet Configuration

cpipero, 2019-01-07:
Off of the pubsub branch.

I am trying to run the sample server. I can get the connection to the broker working and I think I get the _connection->writerGroup->writer_ configuration idea (by studying Part 14). But the `DataSetWriter` obviously needs a `DataSet` as shown in this screenshot
[image]

So I proceed to try configuring a `Publisher DataSet`, 
[image]

but when I try to apply it fails.

The code breaks as it seems I am missing a step to prepare the necessary structure (or adding the wrong variables, although i tried different ones). I hope it's just me not being able to understand all this from source.
[image]

Thanks
Cos

rudoq32, 2019-06-18:
Hi @cpipero , 
I am dealing with same issue. 
I made the group box visible by "Convert to Local Variable" (because I am not so much familiar with wpf), but it did not solve my issue. Did I make the fix wrong way?

cpipero, 2019-06-18:
I actually did not fix it, but I found out that if you right click in the blank area the button appears. The _PublisherHeader_ is visible on the top (in debug mode you get the black bar but you can still see it)