OPCFoundation/UA-.NETStandard #626: InstanceNode.IsEqual returns true on unequal nodes

AeonSake, 2019-01-08:
The `IsEqual` method on the InstanceNode (and furthermore their derived classes like ObjectNode, VariableNode, etc) doesn't check the base class `IsEqual` method and always returns true even if the Nodes themself are not equal.

```csharp
/// <summary cref="IEncodeable.IsEqual(IEncodeable)" />
public override bool IsEqual(IEncodeable encodeable)
{
    if (Object.ReferenceEquals(this, encodeable))
    {
        return true;
    }

    InstanceNode value = encodeable as InstanceNode;

    if (value == null)
    {
        return false;
    }

    return true;
} 
```
`Opc.Ua.DataTypes.cs:18345`

Is that intentional, and if so, why do other `IsEqual` methods reference their base-classes but this doesn't?

AlinMoldovean, 2019-01-30:
Hi @AeonSake,

This seems to be an issue in the generated code imported from the [UA NodeSet](https://github.com/OPCFoundation/UA-Nodesetl) repository

AlinMoldovean, 2020-11-06:
Issue reported in [UA-ModelCompiler/ #65](https://github.com/OPCFoundation/UA-ModelCompiler/issues/65)

mrsuciu, 2021-09-14:
The issue is fixed

closed 2021-09-14 by mrsuciu