OPCFoundation/UA-.NETStandard #627: Opc-ua getting started help

sommmen, 2019-01-14:
Hi guys,

I am trying to create my own opc-ua server but so far i am a bit overwhelmed by this sdk.
I already saw some other issues calling for more extended documentation for starters (like myself) e.g. #440 .
I was hoping that someone could provide me with some steps as how to approach building my own opc-ua server.

In my particular case i just need:
- A string node
- A boolean node
Authentications and other matters are for a second stage.

I browsed the code a bit (It is structured/documented quite well - gj!) and i gues i need to create my own SampleServer class and generate the nodes/structures with the https://github.com/OPCFoundation/UA-ModelCompiler

Could someone lay out the steps in short - i will ofcourse investigate more myself but i could do with a bit of a task list of stuff to look at.

Thanks for any efforts in advance.

astrowalker, 2019-01-15:
You don't have to use modeler -- you can code everything directly in C# as the samples show. You create server, add some nodes, update them or not and voila.

sommmen, 2019-01-15:
> You don't have to use modeler -- you can code everything directly in C# as the samples show. You create server, add some nodes, update them or not and voila.

Thanks for your reply.

Let me explain a bit more of what i'm looking for.
http://documentation.unified-automation.com/uasdkcpp/1.6.2/html/L2TutorialServerHelloWorld.html

This has a list of steps; set configuration, do this in the server, then do that.
I'm looking for something like this.

astrowalker, 2019-01-15:
Did you look at: http://opcfoundation.github.io/UA-.NETStandard/help/index.htm#server_development.htm

Of course the layout is different and it is an overview -- in order to fully understand what is going on, you have to open the code in Visual Studio and check the code. For such basics things IMHO it is pretty straightforward. 

I suggest donwloading UA-Expert and connecting to sample server, after you succeed try to change something in sample server -- add a node, or change the type of existing one, and run it again and check if UA Expert sees the change.

sommmen, 2019-01-16:
I can run both examples.

I am now trying to just host a ua:string variable in the sample-server.
in the end i would like something like:

I see that in the examples the first boiler is defined in predefinedNodes.xml, and the second one is created with CreateBoiler() in CreateAdressSpace() (class BoilerNodeManager).

So far the right steps would be copying (most of) the sample server code,
And add my own 'node manager'.

I am now in the process of figuring out where exactly the boiler is created, and where the variables are being set/get so that i can use that knowledge for my own case.

AlinMoldovean, 2019-01-21:
@sommmen ,

You can use the Reference Server sample as a starting point.
For writing values (for the client side) check the implementation of OnWriteDataItem() or OnWriteAnalog() methods in ReferenceNodeManager class.
For reading values, you can have a loop that sets the value of the nodes, just like the DoSimulation() method ReferenceNodeManager class.

NakWarsi, 2019-01-23:
@sommmen 
exploring OPC UA, I recently Created very minimal CLient and Server (console Application) in .Net with the help of OPC foundation's reference server,
You can follow the reference server to get the Clear Idea as @AlinMoldovean  have explained or if you need the exact example than you can explore https://github.com/NakWarsi/OPC_SERVER_DotNet_Explorer,

sommmen, 2019-01-24:
Thanks @AlinMoldovean and @NakWarsi (and @astrowalker) for your help. I managed to use the opcuamodeler tool to create a bit of a test setup. your project @NakWarsi was kind of exactly what i was looking for. I left this question open because i kind of wanted to present my sample but unfortunately for now i lack the time. I think this post will however give newcomers the right info.

closed 2019-01-24 by sommmen