OPCFoundation/UA-.NETStandard #630: Opc.Ua.ServiceResultException

NakWarsi, 2019-01-21:
I am trying to create a Client that console client application is running fine when i am running in console ( by dotnet run command) but when i am running it in visual studio it's throwing two exception ("Opc.Ua.ServiceResultException" and "System.AggregateException"),
can't figureout why is this happening,

OUT PUT
`'dotnet.exe' (CoreCLR: DefaultDomain): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\2.1.6\System.Private.CoreLib.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'dotnet.exe' (CoreCLR: clrhost): Loaded 'C:\Users\Wars_Nau\source\repos\OPC_UA\ClientTest1\ClientTest1\bin\Debug\netcoreapp2.1\ClientTest1.dll'. Symbols loaded.
'dotnet.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\2.1.6\System.Runtime.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'dotnet.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\2.1.6\System.Console.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'dotnet.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\2.1.6\System.Threading.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'dotnet.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\2.1.6\System.Runtime.Extensions.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'dotnet.exe' (CoreCLR: clrhost): Loaded 'C:\Users\Wars_Nau\.nuget\packages\opcfoundation.netstandard.opc.ua\1.4.354.23\lib\netstandard2.0\Opc.Ua.Core.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'dotnet.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\2.1.6\netstandard.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'dotnet.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\2.1.6\System.Threading.Tasks.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'dotnet.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\2.1.6\System.Collections.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'dotnet.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\2.1.6\System.Net.NameResolution.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'dotnet.exe' (CoreCLR: clrhost): Loaded 'C:\Users\Wars_Nau\.nuget\packages\opcfoundation.netstandard.opc.ua\1.4.354.23\lib\netstandard2.0\Opc.Ua.Client.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'dotnet.exe' (CoreCLR: clrhost): Loaded 'C:\Users\Wars_Nau\.nuget\packages\opcfoundation.netstandard.opc.ua\1.4.354.23\lib\netstandard2.0\Opc.Ua.Configuration.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'dotnet.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\2.1.6\System.IO.FileSystem.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'dotnet.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\2.1.6\System.Runtime.Serialization.Xml.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'dotnet.exe' (CoreCLR: clrhost): Loaded 'C:\Program Files\dotnet\shared\Microsoft.NETCore.App\2.1.6\System.Private.DataContractSerialization.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Exception thrown: 'Opc.Ua.ServiceResultException' in System.Private.CoreLib.dll
Exception thrown: 'System.AggregateException' in System.Private.CoreLib.dll
The program '[816] dotnet.exe' has exited with code 130 (0x82).
`
please let me knowif any one has any idea

AlinMoldovean, 2019-01-21:
@NakWarsi ,

Could you put a complete stack trace for the exception?

NakWarsi, 2019-01-22:
call Stack:-
>	ClientTest1.dll!ClientTest1.MyReferenceClient.ConsoleClient() Line 97	C#
 	ClientTest1.dll!ClientTest1.MyReferenceClient.Run() Line 59	C#
 	ClientTest1.dll!ClientTest1.Program.Main(string[] args) Line 32	C#

if i am running the program than it is throwing exception while loading config file but if i am running it with dotnet run (in console) it's running fine,

code:-`using System;
using System.Net;
using System.Threading;
using System.Threading.Tasks;
using Opc.Ua;
using Opc.Ua.Client;
using Opc.Ua.Configuration;
using System.Collections.Generic;
using System.Diagnostics;

namespace ClientTest1
{
    //exit codes to return on status
    public enum ExitCode : int
    {
        Ok = 0,
        ErrorClientNotStarted = 0x80,
        ErrorClientRunning = 0x81,
        ErrorClientException = 0x82,
        ErrorInvalidCommandLine = 0x100
    };

    class Program
    {
        public static int Main(string[] args)
        {
            Console.WriteLine(".Net Core OPC UA Testing Client");
            bool autoAccept = false;

            MyReferenceClient Client = new MyReferenceClient(autoAccept);
            Client.Run();

            return (int)MyReferenceClient.ExitCode;
        }
    }

    public class MyReferenceClient
    {
      //  ClientOPC client;
        Task status;
        static bool autoAccept = false;
        static ExitCode exitCode;

        public MyReferenceClient(bool _autoAccept)
        {
            autoAccept = _autoAccept;
        }

        // returning Status of the Client 
        public static ExitCode ExitCode { get => exitCode; }

        // running the Client
        internal void Run()
        {
            try
            {
                exitCode = ExitCode.ErrorClientNotStarted;
                ConsoleClient().Wait();
                Console.WriteLine("Client started. Press Ctrl-C to exit...");
                exitCode = ExitCode.ErrorClientRunning;
            }
            catch (Exception ex)
            {
[... 139 lines cut ...]
                //}

                list.ForEach(i => i.Notification += OnNotification);
                subscription.AddItems(list);
                
                //adding subscription to the session
                session.AddSubscription(subscription);
                subscription.Create();
                Console.WriteLine(subscription.NotificationCount);

            }
        }
        private static void OnNotification(MonitoredItem item, MonitoredItemNotificationEventArgs e)
        {
            foreach (var value in item.DequeueValues())
            {
                Console.WriteLine("data recieved");
                //Console.WriteLine("{0}: {1}, {2}, {3}, {4}", item.DisplayName, value.Value, value.SourceTimestamp, value.StatusCode,value.StatusCode);
            }
        }
    }
}`

screenshot of running project with dotnet run:-
[https://imgur.com/a/khzOlGd](url)

NakWarsi, 2019-01-22:
there was a problem in session,
It was getting Killed,
now problem solved,

anyway thank you @AlinMoldovean

closed 2019-01-22 by NakWarsi

viktorsolik, 2020-11-11:
Hi, I have a simmilar problem. Can I ask why it was getting killed ? and how did you fixed it ?

Thank you.