OPCFoundation/UA-.NETStandard #632: Reading: session.read() takes 200ms for reading a node

SanderV94, 2019-01-27:
Hi

I am making an application where I will interact with a PLC (S7-1500). When I am reading only one node the request takes 200ms. While when I use another OPC library it only takes <10 ms, when the node is registerd.

` 
var nodes = new List<NodeId>()
            {
                _Barcode1.NodeId,
                _Barcode2.NodeId,
                _Barcode3.NodeId,
                _Barcode4.NodeId
            };

            var types = new List<Type>()
            {
                typeof(string),
                typeof(string),
                typeof(string),
                typeof(string)

            };
            List<object> resultt;
            List<ServiceResult> serviceResults;
            _session.ReadValues(nodes,types, out resultt, out serviceResults);
            time.Stop();
            Console.WriteLine(time.ElapsedMilliseconds);
`
Is there some way to read to nodes quicker from the server?
Thank you

mutsumu-nomura, 2019-01-28:
We had problem with the latest Simatic.net (V14.0) OPC UA server, that MaxAge value in session.Read( ) affetcs the performance. (Server did not get the latest value from PLC but returns cached value, so we had delay of getting fresh values, or timeout, when PLC was not previously connected)

Since client.ReadValue( ) or client.ReadValues( ) are using default value of MaxAge, which does not fit our need, we had to make our own version of (copy&edit versoin of) ReadValue and Read Values with MaxAge = 0.

(You may have different issues. but hope it helps.)

SanderV94, 2019-01-28:
@AlinMoldovean First I was also thinking this must be on the server side but when I use another library and I don't change the hardware. I get 40ms for reading an unregisterd node and <10 ms for registerd nodes. I think that I will use the not official library for client applications, for server I will still stick with the official one. Unless somebody knows a property that is causing that + 200 ms reading delay.

@mutsumu-nomura  thank you for you information, I am not having this issue right now. The s7-1500 acts a server in my setup, so I am reading the values directly from the PLC. This way I don't have any connection problems between client => OPC servr => PLC. My setup is client => PLC (with OPC server)

ThomasLevering, 2019-02-04:
I have not this extra delay, CPU 1515 v2.6
<10 ms for registerd nodes one Item
and 5Sec for 1000 Mixed Items Bool,Int,Real,Array

registerd nodes are 20x faster

with Firmware before 2.6 registerd nodes are not working in subscriptions.

I asked for support 2017-07 and now ther are working, with the same slow speed then normal nodes (add/remove to subscription)

You can use the Siemens System Function "RT_Info" in Mode 11 to Read the Communication Load.

AlinMoldovean, 2019-10-28:
Issue related to Siemens embedded UA servers

closed 2019-10-28 by AlinMoldovean