OPCFoundation/UA-.NETStandard #634: ProgramStateMachineState support transitions not correctly

Hevgen, 2019-02-02:
Class ProgramStateMachineState has wrong and missed entries at s_CauseMappings array:
1. RunningToReady is caused by Reset() method but it should be Internal caused only.
2. SuspendedToReady is caused by Reset() method but it should be Internal caused only.
3. ReadyToHalted transition is missed but should be caused by Halt() method.

My opinion is based on OPC Unified Architecture Specification Part 10: Programs (Release 1.04).

EthanChangAED, 2023-08-04:
Hello @Hevgen, are you still seeing this issue?

Hevgen, 2023-08-06:
Hello. 
yes. All three sub-issues exist.
Transition ReadyToHalted exists now but has no cause in ProgramStateMachineState.s_CauseMappings (should be caused by Methods.ProgramStateMachineType_Halt).

My opinion is based on Part 10: Programs. p.4.2.3 Program Finite State Machine (release 1.05)

closed 2024-12-03 by mregen