OPCFoundation/UA-.NETStandard #635: trouble with server certificate (certificate is not trusted)

NakWarsi, 2019-02-04:
exploring OPC UA I developed Client and Server those are able to communicate but if I am trying to make a connection between my own server and the reference Client of this Repo i get the the exception "untrusted certificate"
full call stack is as given below
[image]

Please help
Thanks

AlinMoldovean, 2019-02-04:
@NakWarsi ,

First you need to manually establish the trust relationship of application instance certificates.

NakWarsi, 2019-02-04:
@AlinMoldovean 
As I am very new to OPC UA  I still didn't get you.
can you refer a document, reference code or anything where i can understand how I can manually establish the trust relationship of application instance certificates.

NakWarsi, 2019-02-04:
thankyou

closed 2019-02-04 by NakWarsi

gustinmi, 2020-07-31:
You modify your code     `private static void CheckCertificateDomain(ConfiguredEndpoint endpoint)` to return true always, and add    `<AutoAcceptUntrustedCertificates>true</AutoAcceptUntrustedCertificates>` into config of the client. 
This is just for developing and troubleshooting of course. The reference implementation is what the name suggest : REFERENCE.