OPCFoundation/UA-.NETStandard #640: Four NamespaceIndexes in .XML file in Visual studio

PrudhviRajC, 2019-02-11:
Hello,
I am new to OPC UA, I am trying to implement the OPC UA Server in Visual Studio by using the Lesson 1 and Lesson 2 tutorials of the Unified Automation website. The tutorials have a sample server with only one namespace as shown below, also it is described that it is a "DefaultNamespaceIndex"

The example described in the tutorial has one namespace and so in code it was assigned to DefaultNamespaceIndex as shown

`<NamespaceUris>
    <Uri>http://opcfoundation.org/UA/AML/</Uri>
 </NamespaceUris>
`
public override void Startup()
        {
            try
            {
                Console.WriteLine("Starting AMLOneNodeManager.");

                DefaultNamespaceIndex = AddNamespaceUri("http://opcfoundation.org/UA/AML/");

                Console.WriteLine("Loading the AMLOne Model.");
                ImportUaNodeset(Assembly.GetEntryAssembly(), "DeviceModelling.xml");
            }
            catch (Exception e)
            {
                Console.WriteLine("Failed to start AMLOneNodeManager " + e.Message);
            }
        }

But I have Information model (.XML file) with two namespaces, does anyone know how to program in C# .i.e., if there exists more than one namespaceIndex?

`  <NamespaceUris>
    <Uri>http://opcfoundation.org/UA/AML/</Uri>
    <Uri>http://www.xyz.de/DeviceModelling.aml</Uri>
  </NamespaceUris>`

Thank youin advance.

NichtEuler, 2019-02-13:
Are you using their SDk or this?
I guess you are using theirs.
If you are using the Unified Automation SDK I would recommend asking them.
Otherwise I would like to point you to #546

PrudhviRajC, 2019-02-13:
Thanks for the message, I am using the Unified Automation SDKs.
Also, I am thinking that can we use the xml file (from AutomationML editor)which was created outside the Unified Automation UaModeler can be used in the .NET SDK C# code?

AlinMoldovean, 2019-02-13:
@PrudhviRajC ,

Please have a  look at #546

closed 2019-02-13 by AlinMoldovean