OPCFoundation/UA-.NETStandard #641: Thread Safety of NodeState classes

sommmen, 2019-02-11:
Dear All,

I created a custom node manager, with 3 nodes and each of them has some variables (string, bool).
Now i'd like to use a seperate thread that updates information on the node every second or so.

I used the opc-ua modeler to create custom node classes. These all simply inherit from BaseObjectState.
Now I am writing to an instance of this object every second (MyObject.TestVariable.Value = "ATestValue"), and the client may read the value whenever. 

Can I assume this is thread safe, or could the read/write operations cause an error?

Thanks in advance.

fchaxel, 2019-02-13:
Hi @sommmen 

On CustomNodeManager2 an object is available for that : Lock.

So lock (Lock) {  ... }

Bye.

sommmen, 2019-02-13:
> Hi @sommmen
> 
> On CustomNodeManager2 an object is available for that : Lock.
> 
> So lock (Lock) { ... }
> 
> Bye.

Hello, and thanks for helping.

Right now, I run a thread and pass the opcua node as a reference. In the Thread it will do some stuff to the opcua node (read out a bool, write a string variable etc). My concern is that the Client fills the node object (referenced in my thread) and then my server thread will also read/write creating an error. Would Using the lock in my thread also stop the client from writing/reading the opcua node and thus solve my problem?

I cannot find direct evidence in the customnodemanager2 that supports this claim, but will investigate further.

AlinMoldovean, 2019-02-13:
@sommmen ,

You need to use the Lock of the CustomNodeManager2 instance that holds your node.

sommmen, 2019-02-14:
> @sommmen ,
> 
> You need to use the Lock of the CustomNodeManager2 instance that holds your node.

Thanks for your attention aswell. Does this address my concern below?

> Hello, and thanks for helping.
>
> Right now, I run a thread and pass the opcua node as a reference. In the Thread it will do some stuff to the opcua node (read out a bool, write a string variable [...]

closed 2019-03-25 by AlinMoldovean