OPCFoundation/UA-.NETStandard #644: Information Model in XML

NakWarsi, 2019-02-12:
being an OPC UA explorer, I was searching for a way where i can define my information model in XML or JSON format (file) and our OPC UA server can consume that only so i don't have to change the code of my server's nodemanager file again and again whenever i need to change my information model.
in this way i have to write just one server for all kind of information model.

is it possible ?
if yes how ? 
is there any sample available ?

any concern will be appreciated.
Thank You

fchaxel, 2019-02-13:
Hi @NakWarsi 

Get the model compiler and use it to generates code from an XML nodeset description. 

https://github.com/OPCFoundation/UA-ModelCompiler

Have a look to #546 and #563 

Bye

mpostol, 2019-02-13:
Have a look on the document to get more about UA IM deployment:

- [OPC UA Information Model Deployment][CAS.OPCUAIMD]

[CAS.OPCUAIMD]: http://www.commsvr.com/InternetDSL/commserver/P_DowloadCenter/P_Publications/20140301E_DeploymentInformationModel.pdf

closed 2019-02-14 by NakWarsi

mpostol, 2019-02-14:
@NakWarsi there is project facing information representation in the UA environment:
[Object Oriented Internet](https://github.com/mpostol/OPC-UA-OOI)
Consider joining and continue the discussion on [Gitter]( https://gitter.im/mpostol/OPC-UA-OOI)

NakWarsi, 2019-02-14:
Thankyou @mpostol  that project e really really interesting
and thanyyyouffnitng me i Gitter room to hoping for the good learning in future