OPCFoundation/UA-.NETStandard #648: problem with Information XML Model

NakWarsi, 2019-02-14:
In my server I am giving the Information model in XML format server is running fine but i can't see the node i have created.
My XML file:-
````
<?xml version="1.0" encoding="utf-8"?>
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:s1="http://yourorganisation.org/testBlaBla2/Types.xsd" xmlns:ua="http://unifiedautomation.com/Configuration/NodeSet.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema">
    <NamespaceUris>
        <Uri>http://yourorganisation.org/testBlaBla2/</Uri>
    </NamespaceUris>
    <Aliases>
        <Alias Alias="Double">i=11</Alias>
        <Alias Alias="Organizes">i=35</Alias>
        <Alias Alias="HasTypeDefinition">i=40</Alias>
    </Aliases>
    <Extensions>
        <Extension>
            <ua:ModelInfo Tool="UaModeler" Hash="P3NrNnM59BU3S10wNAdYQw==" Version="1.6.2"/>
        </Extension>
    </Extensions>
    <UAObject SymbolicName="BaseObject" NodeId="ns=1;i=5002" BrowseName="1:BlaBla">
        <DisplayName>BlaBla</DisplayName>
        <Description>BlaBlaTEsting</Description>
        <References>
            <Reference ReferenceType="Organizes" IsForward="false">i=85</Reference>
            <Reference ReferenceType="Organizes">ns=1;i=5003</Reference>
            <Reference ReferenceType="Organizes">ns=1;i=6001</Reference>
            <Reference ReferenceType="Organizes">ns=1;i=5001</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=61</Reference>
        </References>
    </UAObject>
    <UAObject NodeId="ns=1;i=5003" BrowseName="1:dewdededwefwed">
        <DisplayName>dewdededwefwed</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=61</Reference>
            <Reference ReferenceType="Organizes" IsForward="false">ns=1;i=5002</Reference>
        </References>
    </UAObject>
    <UAVariable DataType="Double" NodeId="ns=1;i=6001" BrowseName="1:eddewdew22222" AccessLevel="3">
        <DisplayName>eddewdew22222</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
            <Reference ReferenceType="Organizes" IsForward="false">ns=1;i=5002</Reference>
        </References>
    </UAVariable>
    <UAObject NodeId="ns=1;i=5001" BrowseName="1:edweded">
        <DisplayName>edweded</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=58</Reference>
            <Reference ReferenceType="Organizes" IsForward="false">ns=1;i=5002</Reference>
        </References>
    </UAObject>
</UANodeSet>
````

Server's Node Manager Code :-

````
public override void CreateAddressSpace(IDictionary<NodeId, IList> externalReferences)
{
lock (Lock)
{
            IList<IReference> references = null;

            if (!externalReferences.TryGetValue(ObjectIds.ObjectsFolder, out references))
            {
                externalReferences[ObjectIds.ObjectsFolder] = references = new List<IReference>();
            }

            string resourcepath = "C:/Users/Wars_Nau/Desktop/OPC/ServerTest2/ServerTest2/testblabla2.xml";
            ImportXml(externalReferences, resourcepath);
        }
    }

    private void ImportXml(IDictionary<NodeId, IList<IReference>> externalReferences, string resourcepath)
    {
        try
        {

            NodeStateCollection predefinedNodes = new NodeStateCollection();

            Stream stream = new FileStream(resourcepath, FileMode.Open);
            UANodeSet nodeSet = UANodeSet.Read(stream);

            SystemContext.NamespaceUris.Append(nodeSet.NamespaceUris.ToString());
            nodeSet.Import(SystemContext, predefinedNodes);

            for (int ii = 0; ii < predefinedNodes.Count; ii++)
            {
                AddPredefinedNode(SystemContext, predefinedNodes[ii]);
            }
            // ensure the reverse refernces exist.
            AddReverseReferences(externalReferences);
        }
        catch (Exception)
        {

            throw;
        }
    }
````
i suppose to get the Node Hierarchy like this as i created with unified automation's GUI software

[image]

but I am getting this

[image]

I can't figure out where i am lacking or missing at.
please help.

Thank You

AlinMoldovean, 2019-02-14:
@NakWarsi ,

Check if the http://yourorganisation.org/testBlaBla2/ namespace is defined n your node manager.

NakWarsi, 2019-02-14:
yeah it wasn't defined.
Thank You very much @AlinMoldovean

closed 2019-02-14 by NakWarsi

johnjoal, 2019-02-15:
According to the code, AddPredefinedNode should also add types defined in xml, but I can't make it work.
For example this xml below, with a simple object type, I can't see it is added to the servers ObjectTypes.

```<?xml version="1.0" encoding="utf-8"?>
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" 
           xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" 
           xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" 
           xmlns:ua="http://unifiedautomation.com/Configuration/NodeSet.xsd" 
           xmlns:xsd="http://www.w3.org/2001/XMLSchema">
    <NamespaceUris>
        <Uri>http://myorg.org/UA/</Uri>
    </NamespaceUris>
    <Aliases>
        <Alias Alias="Int16">i=4</Alias>
        <Alias Alias="UInt32">i=7</Alias>
        <Alias Alias="Organizes">i=35</Alias>
        <Alias Alias="HasTypeDefinition">i=40</Alias>
        <Alias Alias="HasProperty">i=46</Alias>
    </Aliases>
    <UAObjectType BrowseName="1:BufferType" NodeId="ns=1;i=4002">
      <DisplayName>BufferType</DisplayName>
      <Description>BufferType</Description>
      <References>
        <Reference IsForward="false" ReferenceType="HasSubtype">i=58</Reference>
      </References>
    </UAObjectType>
</UANodeSet>