OPCFoundation/UA-.NETStandard #650: Certificate validation for chains requires CA certificate to be trusted

ThomasNehring, 2019-02-15:
According to the spec, part 4 section 6.1.3 (rel. 1.03 or 1.04) certificate validation (e.g. validation of server certificate on client side) requires that

1.  the whole certificate chain of the server is known to the client 
2.  all certificates are valid and the signatures can be validated
3.  at least one certificate in the chain is trusted (i.e. has been placed into the trust store).

The following is therefore a valid configuration (assuming 2. is fulfilled and assuming a certificate chain of length 2): 

the application certificate of the server is in the clients trust store, while the the CA certificate is in the issuer store.

In this scenario, however, certificate validation fails with the error message "Certificate issuer is not trusted". The message itself is correct (the CA certificate is not in the trust store), but it is not a reason to let validation fail, since the application certificate is trusted.

The reason for the issue seems to be the coding in CertificateValidator.cs, line 792 and subsequent.

In the lines before that point the certificate chain has been constructed correctly. Then there is an if clause

if(issuedByCA && !isIssuerTrusted)

which correctly evaluates to true. Then, however, there is another if clause

if(m_applicationCertificate == null || !Utils.IsEqual(m_applicationCertificate.RawData, certificate.RawData))

which, in my opinion, does not make sense. At runtime it can be seen in the debugger that "m_applicationCertificate" is the client certificate (here it is assumed that a client tries to validate a server chain), while "certificate" holds the servers application certificate. It does not make sense to compare these two. What has to be checked here is whether the servers application certificate is trusted, since it already known that this is not true for the issuer.

mregen, 2019-02-21:
Hi @ThomasNehring, I agree with your finding that the code snippet should be removed. Please provide a Pull request. However, @AlinMoldovean, since this is at the core of cert validation we should run a CTT pass to double check we are not opening a security hole.
Thanks,
Martin

ThomasNehring, 2019-02-21:
Hello  <mailto:Martin.Regen@microsoft.com> @Martin Regen, 

 

this may take a while, as we don’t have a fix for this right now. Just removing the code snippet will, very likely not work. Actually we fixed a similar issue in the Legacy stack, and back then I did not just delete this snippet, very likely for good reason. But since that fix from us is not visible in the .Net standard stack there might have been issues with that fix when you migrated to .Net standard(?). I can check whether it would work here, though.

 

Best regards,

 

Thomas

AlinMoldovean, 2019-03-22:
Fixed in #679

closed 2019-03-22 by AlinMoldovean