OPCFoundation/UA-.NETStandard #655: Node.Write method fails on writing attributes which aren't of type DataValue

AeonSake, 2019-02-18:
The `Write` method on the Node class (`ILocalNode` interface) fails to write any attribute which aren't of type DataValue. If I understand the idea of that method correctly, it should provide a generalized access to the attributes of a node. But in the current state it only allows writing attributes which are from type DataValue (and there are currently no attributes matching this criteria).

```csharp
public ServiceResult Write(uint attributeId, DataValue value)
{
    if (!SupportsAttribute(attributeId))
    {
        return StatusCodes.BadAttributeIdInvalid;
    }
            
    // check for read only attributes.
    switch (attributeId)
    {
        case Attributes.NodeId:
        case Attributes.NodeClass:
        {
            return StatusCodes.BadNotWritable;
        }
    }

    // check data type.
    if (attributeId != Attributes.Value)
    {
        if (Attributes.GetDataTypeId(attributeId) != TypeInfo.GetDataTypeId(value))
        {
            return StatusCodes.BadTypeMismatch;
        }
    }

    return Write(attributeId, value.Value);
}
```
`Node.cs:293`

I believe the problem is within the `TypeInfo.GetDataTypeId` method. It misses a check for the DataValue type or there should be a new overload method `TypeInfo.GetDataTypeId(value : DataValue)`.
My suggestion would be adding a new overload since there are already overloads for Type and TypeInfo.
```csharp
public static NodeId GetDataTypeId(DataValue value)
{
    if (value == null)
    {
        return Opc.Ua.NodeId.Null;
    }

    return GetDataTypeId(value.Value);
}
```
`TypeInfo.cs:233`

Please correct me if my thoughts were wrong.

AlinMoldovean, 2019-04-03:
@AeonSake ,

The correct way of writing node attributes in a server is the one from the implementation of `CustomNodeManager2 `class. i.e. by using the `NodeState.WriteAttribute()` method.

```
        /// <summary>
        /// Writes the specified attribute value.
        /// </summary>
        /// <param name="context">The context for the current operation.</param>
        /// <param name="attributeId">The attribute id.</param>
        /// <param name="indexRange">The index range.</param>
        /// <param name="value">The value.</param>
        /// <returns>
        /// An instance of the <see cref="ServiceResult"/> containing the status code and diagnostic info for the operation.
        /// ServiceResult.Good if successful. Detailed error information otherwise.
        /// </returns>
        public ServiceResult WriteAttribute(
            ISystemContext context,
            uint attributeId,
            NumericRange indexRange,
            DataValue value)
        {
            // check for bad parameter.
            if (value == null)
            {
                return StatusCodes.BadStructureMissing;
            }

            object valueToWrite = value.Value;

            if (attributeId == Attributes.Value)
            {
                // writes to server timestamp never supported.
                if (value.ServerTimestamp != DateTime.MinValue)
                {
                    return StatusCodes.BadWriteNotSupported;
                }

                // call implementation.
                try
                {
                    return WriteValueAttribute(
                        context,
                        indexRange,
                        valueToWrite,
                        value.StatusCode,
                        value.SourceTimestamp);
                }
                catch (Exception e)
                {
                    return new ServiceResult(e, StatusCodes.BadUnexpectedError);
                }
            }

            // writes to status code or timestamps never supported.
            if (value.StatusCode != StatusCodes.Good || value.ServerTimestamp != DateTime.MinValue || value.SourceTimestamp != DateTime.MinValue)
            {
                return StatusCodes.BadWriteNotSupported;
            }

            // cannot use index range for non-value attributes.
            if (indexRange != NumericRange.Empty)
            {
                return StatusCodes.BadIndexRangeInvalid;
            }

            // call implementation.
            try
            {
                return WriteNonValueAttribute(context, attributeId, value.Value);
            }
            catch (Exception e)
            {
                return new ServiceResult(e, StatusCodes.BadUnexpectedError);
            }
        }
```

AeonSake, 2019-04-04:
I'm talking about client-side. When reading the attributes of a node, I need to set all the attributes myself with the `Session.Read()` result. While I currently set each individual attribute I was questioning the purpose of the `ILocalNode.Write()` method. It seemed to me as if this method was intended for that use.

Example:
```csharp
var attributesToRead = new ReadValueIdCollection();
attributesToRead.Add(new ReadValueId {NodeId = node.NodeId, AttributeId = Attributes.Description}); //just as an example

Session.Read(
  null,
  0,
  TimestampsToReturn.Server,
  attributesToRead,
  out var res,
  out _);
ClientBase.ValidateResponse(res, attributesToRead);

if (res[0] != null && res[0] is LocalizedText description) node.Description = description;
```
This works but gets a bit clustered when reading multiple attributes. Also, all validations have to be done by the client.  
The `ILocalNode.Write()` method seems to do exactly that: validate attribute values and write if valid. But as in my original post, it misses a critical check and always fails.
```csharp
//usage with 'ILocalNode.Write()'
if (res[0] != null) node.Write(Attributes.Description, res[0]);
```

closed 2019-11-25 by AlinMoldovean

AeonSake, 2021-03-23:
I want to note that this issue still exists in the latest version. `Node.Write()` is absolutely useless on the client-side with the current implementation.